nightscout/AndroidAPS #3153: strange format of basal calculation

idalioreis, 2024-01-01:
Hi, 

I am facing a problem with aps, Its on dynisf and no smooth.
The number are with strange format.
If i put openaps SMB algorithm work fine.

- dev 3.3.0-dev
[image]
[image]
[AndroidAPS_LOG_1704130454663.log.zip](https://github.com/nightscout/AndroidAPS/files/13806032/AndroidAPS_LOG_1704130454663.log.zip)

MilosKozak, 2024-01-01:
please locate logs from time on screenshot

idalioreis, 2024-01-01:
logs_2024-01-01 17.34

MilosKozak, 2024-01-02:
one more time

closed 2024-01-12 by MilosKozak

fredrik-ljung, 2024-01-13:
Hi.
We are experiencing this error. The ISF value is printed like in the picture above every time Dynamic ISF is activated. OpenAPS SMB algorithm works fine.
It started after an execution of Autotune.
I unfortunately don't have logs from the time when it stared but I will try to provide logs from an attempt to activate the Dynamic ISF algorithm.
Is there a solution to the problem? How can we get it running again?

Currently using a one week old version of the dev branch.

fredrik-ljung, 2024-01-13:
Used version 77b622ba08e4f3b5358b9c4e54d4e2503d4c3add.
[AndroidAPS_LOG_1705162401737.log.zip](https://github.com/nightscout/AndroidAPS/files/13928790/AndroidAPS_LOG_1705162401737.log.zip)

fredrik-ljung, 2024-01-14:
I found out what is causing the immediate problem, but not the root cause of the problem. For some reason the ApsDynIsfAdjustmentFactor was reset to 0, which in turn gave incorrect sensitivity values and 0 as TDD value.
I have no idea why the value was changed to 0. We have not touched it and 0 is an illegal value.
After setting a proper ApsDynIsfAdjustmentFactor value the Dynamic IFS algorithm works as expected again.
I have tried but not been able to reproduce the error.