nightscout/AndroidAPS #3154: Steps Graph should be simplified

Philoul, 2024-01-02:
I can't understand steps graph, with lots of overlapping on some periods

[image]

In above graph, I have bars close to 40 and at the same time a green bar close to 180 ?

To be able to understand graph, I think we should show for example something like "number of steps during the last minute" to avoid overlapping...

Philoul, 2024-01-09:
additional screenshot during sport training
[image]

HR should be smoothed but we see the increase duting training and decrease after.
Concerning steps, it's a mess...

DM-LTR, 2024-01-10:
It would be nice to have steps in form of bar graph, the same as deviations

PR #3369 (2024-06-18): Fix Overlapping in Step Graph
Fix #3154
Step duration 5min and graph duration was previously 15min

This PR use same duration for graph
=> Overlapping is not completely removed (within watch we have one measure every 40s but it improves graph readings
closed 2024-10-16 by Philoul