nightscout/AndroidAPS #3155: Switching Porfile via NSclient V3 doesn't work with APPS 3.0.2.3 and NS 15.02

gerison77, 2024-01-03:
After creating a new Dataset with new basal rates in NS with profile manager. Switching to this profile in NS is not effected on AAPS 3.0.2.3 app with NSclient V3 enabled unless switching to NSclient2. 

Example: 
Profile switch in NS at 7:02 was effected in AAPS directly after switching to NSclientV1 at 8:35.

[image]
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13816851/AndroidAPS.log)

MilosKozak, 2024-01-03:
log starts 9:06

MilosKozak, 2024-01-03:
i tested it and it works like expected. If you have websockets enabled, change is performed immediately. Without websockets it takes up to 5 min.
I can check your case if you find logs

gerison77, 2024-01-03:
Thank you Milos.
I reproduced the behaviour today at 23:03 (profile switch in NS) than switching to NSclient V1, with effected switch at 23:05.
Websockts were activated in NSClient V3.

MilosKozak, 2024-01-03:
upload logs. best from both sides

gerison77, 2024-01-04:
[AndroidAPS._2024-01-03_21-25-10_.109.zip](https://github.com/nightscout/AndroidAPS/files/13827916/AndroidAPS._2024-01-03_21-25-10_.109.zip)
[AndroidAPS._2024-01-03_21-25-10_.110.zip](https://github.com/nightscout/AndroidAPS/files/13827919/AndroidAPS._2024-01-03_21-25-10_.110.zip)

MilosKozak, 2024-01-04:
i need log from aapsclient too

gerison77, 2024-01-04:
AAPS Client was not involved. Profile switch effected directly on nightscout site, hosted by horoku. Profile switch via AAPS Client works.

MilosKozak, 2024-01-04:
start with reading release notes
https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Releasenotes.html#important-comments-on-using-v3-versus-v1-api-for-nightscout-with-aaps

closed 2024-01-04 by MilosKozak

gerison77, 2024-01-04:
Im about to get the nightscout logs as well.

gerison77, 2024-01-04:
AAPS Client was not involved. But I read now, that entries made in NS Website require NS 1 so. The issue is known.
Sorry for bothering.