nightscout/AndroidAPS #3158: [Feature request]: Enhancement Proposal for Database Cleanup feature in Maintenance section

dar-kod, 2024-01-06:
I would like to suggest a feature in **Maintenance** ⇒ **Database Cleanup** to specify the number of days after which older data can be deleted? Not only do I not need data older than 14 days (or maybe 7), but I think it could significantly increase battery life. I would also suggest that the cleanup can be done automatically, e.g. once a day.

robertrub, 2024-01-06:
Yes, user deciding the number of days to keep is a good option. Should set limits though so that they don't delete too much (in ref to DynISF)

Running it every day might be too much (won't gain a lot of battery by deleting one day's data but the cleaning itself will use battery). Maybe _auto clean_ once a week or every 2 weeks ?

MilosKozak, 2024-01-09:
Running cleanup from maintenance keeps 3 months and removes tracked changes too
Automatic run of of cleanup is scheduled daily and keep 6 months of data
with 3.2 doing more doesn't have significant impact to performance and battery life
if you have 3.1, update!

closed 2024-01-09 by MilosKozak