nightscout/AndroidAPS #3159: Dana RS: Loop gets automatically disabled/enabled

Truman1234, 2024-01-07:
I am running the 3.2.0.2-dev version of AndroidAPS and since about a month I am facing a problem, that the loop gets automatically disabled and then reenabled from time to time.

[image]

When I press the loop icon, it shows this:

[image]

I seems, like there is a bluetooth connection problem, but when I go on the Dana RS tab in AndroidAPS, to read the stats from the pump, it connects to the pump and reads the values.

I am using a Samsung M51 for AndroidAPS, but also tested another android phone with the same result. 

Additionally, I also tried an older version of AndroidAPS, where I never had that problem, but it also shows now the same behavior.

So it seems to be a pump connection issue, but I don´t understand, why it the can easily connect to the pump, when I am reading the stats from my Dana RS via AndroidAPS

Truman1234, 2024-01-07:
And what I forgot: The loop still seems to work, when it is in disabled mode, but on the action tab and at the bottom of the main screen all the pump related buttons are gone.

MilosKozak, 2024-01-07:
When the pump is not initialized Loop status is set to disabled (and you don't see Calc an Insulin button as well)

Truman1234, 2024-01-07:
So you are saying, the pump is faulty ? Or is it a software problem ?

MilosKozak, 2024-01-07:
impossible to say from this one screenshot

Truman1234, 2024-01-07:
What is necessary ? The log file ?

MilosKozak, 2024-01-07:
I see what you tested so I'd expect faulty pump as first option

Truman1234, 2024-01-07:
Ok, I will check back with the Dana representative. Thank you.

closed 2024-01-07 by MilosKozak