nightscout/AndroidAPS #3160: Issue with SMB using xDrip BG broadcasts from G7 sensor following the latest xDrip nightly

kropas13, 2024-01-07:
AAPS no longer accepts the transmitted BG values for SMBs (Safety warning in AAPS/SMB: SMB always and after carbs disabled because active BG source doesn't support advanced filtering), after having updated xDrip+ to the latest nightly (https://github.com/NightscoutFoundation/xDrip/commit/3b418f2cd9b5d5dfaa9fc1b0f506f8a7adcf9a4f-2024.01.05). 

Used software/devices
- AAPS v3.2.0.3 (BG source: xDrip+)
- xDrip+ (as CGM collector)
- Pixel 5
- CGM Dexcom G7 

It looks like the new xDrip broadcast identifier (G7) is the culprit (not identified by AAPS as a reliable BG source), see https://github.com/NightscoutFoundation/xDrip/discussions/3278.

parapenT1sta, 2024-01-07:
G7 needs to be add. Before G7 was identified as G6.

https://github.com/NightscoutFoundation/xDrip/pull/3220/commits/5ef0937cb8cb74c3926427c1c167707c8da6f419

MilosKozak, 2024-01-08:
https://github.com/nightscout/AndroidAPS/commit/44966f3bd842fda40a36eca96c1f393f12dc6102

closed 2024-01-08 by MilosKozak