nightscout/AndroidAPS #3162: Crash preventing an application start after updating from 3.1.0.3 to 3.2.0.3 due to DB migration failure (notes column of boluses table)

OlegShirochenkov, 2024-01-07:
I was using 3.1.0.3, some time ago I have tried a development build on a same device.
Then I've retured back to a 3.1.0.3 - there were some problems with displaying data when downgrading but application have continued to work fine.
A couple of months ago I've upgraded to a 3.1.0.3 with a set of patches for G7 data smoothing.
Today I've decided to upgrade to 3.2.0.3.
Application is crashing right on a start.

According to the log:

```
00:15:27.794 [Thread-10] E/PUMPCOMM: [DanaRSService.readPumpStatus():254]: Unhandled exception
java.lang.IllegalStateException: Migration didn't properly handle: boluses(app.aaps.database.entities.Bolus).
 Expected:
TableInfo{name='boluses', columns={pumpId=Column{name='pumpId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, amount=Column{name='amount', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, utcOffset=Column{name='utcOffset', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, notes=Column{name='notes', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, temporaryId=Column{name='temporaryId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, isValid=Column{name='isValid', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, peak=Column{name='peak', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, type=Column{name='type', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, nightscoutSystemId=Column{name='nightscoutSystemId', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, insulinEndTime=Column{name='insulinEndTime', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, version=Column{name='version', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, pumpType=Column{name='pumpType', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, referenceId=Column{name='referenceId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, isBasalInsulin=Column{name='isBasalInsulin', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, insulinLabel=Column{name='insulinLabel', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, nightscoutId=Column{name='nightscoutId', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, dateCreated=Column{name='dateCreated', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, endId=Column{name='endId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, pumpSerial=Column{name='pumpSerial', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, startId=Column{name='startId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, id=Column{name='id', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=1, defaultValue='undefined'}, timestamp=Column{name='timestamp', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}}, foreignKeys=[ForeignKey{referenceTable='boluses', onDelete='NO ACTION +', onUpdate='NO ACTION', columnNames=[referenceId], referenceColumnNames=[id]}], indices=[Index{name='index_boluses_referenceId', unique=false, columns=[referenceId], orders=[ASC]'}, Index{name='index_boluses_timestamp', unique=false, columns=[timestamp], orders=[ASC]'}, Index{name='index_boluses_pumpId', unique=false, columns=[pumpId], orders=[ASC]'}, Index{name='index_boluses_pumpSerial', unique=false, columns=[pumpSerial], orders=[ASC]'}, Index{name='index_boluses_isValid', unique=false, columns=[isValid], orders=[ASC]'}, Index{name='index_boluses_temporaryId', unique=false, columns=[temporaryId], orders=[ASC]'}, Index{name='index_boluses_pumpType', unique=false, columns=[pumpType], orders=[ASC]'}, Index{name='index_boluses_id', unique=false, columns=[id], orders=[ASC]'}]}
 Found:
TableInfo{name='boluses', columns={id=Column{name='id', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=1, defaultValue='undefined'}, version=Column{name='version', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, dateCreated=Column{name='dateCreated', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, isValid=Column{name='isValid', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, referenceId=Column{name='referenceId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, timestamp=Column{name='timestamp', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, utcOffset=Column{name='utcOffset', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, amount=Column{name='amount', type='REAL', affinity='4', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, type=Column{name='type', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, isBasalInsulin=Column{name='isBasalInsulin', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='undefined'}, nightscoutSystemId=Column{name='nightscoutSystemId', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, nightscoutId=Column{name='nightscoutId', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, pumpType=Column{name='pumpType', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, pumpSerial=Column{name='pumpSerial', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, temporaryId=Column{name='temporaryId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, pumpId=Column{name='pumpId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, startId=Column{name='startId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, endId=Column{name='endId', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, insulinLabel=Column{name='insulinLabel', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, insulinEndTime=Column{name='insulinEndTime', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}, peak=Column{name='peak', type='INTEGER', affinity='3', notNull=false, primaryKeyPosition=0, defaultValue='undefined'}}, foreignKeys=[ForeignKey{referenceTable='boluses', onDelete='NO ACTION +', onUpdate='NO ACTION', columnNames=[referenceId], referenceColumnNames=[id]}], indices=[Index{name='index_boluses_timestamp', unique=false, columns=[timestamp], orders=[ASC]'}, Index{name='index_boluses_referenceId', unique=false, columns=[referenceId], orders=[ASC]'}, Index{name='index_boluses_pumpType', unique=false, columns=[pumpType], orders=[ASC]'}, Index{name='index_boluses_pumpSerial', unique=false, columns=[pumpSerial], orders=[ASC]'}, Index{name='index_boluses_pumpId', unique=false, columns=[pumpId], orders=[ASC]'}, Index{name='index_boluses_temporaryId', unique=false, columns=[temporaryId], orders=[ASC]'}, Index{name='index_boluses_isValid', unique=false, columns=[isValid], orders=[ASC]'}, Index{name='index_boluses_id', unique=false, columns=[id], orders=[ASC]'}]}
	at androidx.room.RoomOpenHelper.onUpgrade(RoomOpenHelper.kt:94)
	at androidx.sqlite.db.framework.FrameworkSQLiteOpenHelper$OpenHelper.onUpgrade(FrameworkSQLiteOpenHelper.kt:253)
	at android.database.sqlite.SQLiteOpenHelper.getDatabaseLocked(SQLiteOpenHelper.java:515)
	at android.database.sqlite.SQLiteOpenHelper.getWritableDatabase(SQLiteOpenHelper.java:413)
	at androidx.sqlite.db.framework.FrameworkSQLiteOpenHelper$OpenHelper.getWritableOrReadableDatabase(FrameworkSQLiteOpenHelper.kt:232)
[... cut ...]
I was trying to perform a backup (adb backup -f APS_Backup.ab -noapk -noobb -noshared info.nightscout.androidaps) hoping to extract a SQLite DB, add a column manually, pack the backup back and restore from it, but it doesn't seem that adb is able to perform such backup without a root privileges. Even when updating to a fullDebug build I was unable to perform such backup.

How is it expected to initiate a boluses table migration in my case?
Currently I've downgraded back waiting for any solution.
In a worst case scenario I will have to reset the database loosing the data (due to a nightscout sync slowness bug in 3.1.0.3).

MilosKozak, 2024-01-08:
Sorry, support is provided only on non modified versions. In your case you did some customisations which led to this error

OlegShirochenkov, 2024-01-08:
migration21to22 (DatabaseModule.kt) was not completely performed by a stable release plan because DB was marked as updated to a higher version while the plan was different and incomplete in a development branch causing an unexpected result.
I've took the master branch (current 3.2.0.3), played with a source code editing: downgraded the DB version to 20 in application, rewrote the migration 20 to 24 plan, then upgraded DB to 24. There were problems with an identityHash but finally, plan was executed and now I'm back on stock stable release 3.2.0.3 from a master branch.
Will not wish to perform this exercise to anyone.

Dear Miloš @MilosKozak , would you be so kind to advise if it's possible to perform the full application backup before trying a dev branch for anything on a live set up on a non-rooted device?

MilosKozak, 2024-01-08:
AAPS can recover database from NS

closed 2024-01-08 by MilosKozak