nightscout/AndroidAPS #3167: Error in CGM inserted since data

robertrub, 2024-01-10:
Latest Dev.

I work with 2 G6 transmitters. Yesterday (Jan 9th at 12:41) I started the 2nd sensor on xDrip variant1 while the first one was at day 9 (on xDrip main). Both xDrip are very recent nightlies.

Sensor 1 ended at 10d (today Jan 10th). I swapped transmitter serials from variant1 to xDrip main. I had to do a "start sensor" sequence just for xDrip to understand that the transmitter was already started. I suppose that sent a "new sensor inserted" message to AAPS.

xDrip variant 1 not being connected to AAPS, yesterday I had manually marked a sensor insertion in AAPS at 12:41 on the 9th.

Now, I remarked that my sensor age was wrong.

[image]

It should be 1 day and 3 hours old.

I deleted the sensor inserted lines from the treatments tab and re-entered a _sensor inserted_ on the 9th at 12:41but the sensor age is still wrong in the main UI.

[image]

[image]

It seems that I can't delete or replace the _new sensor start_ message coming from xDrip.

Here are the logs. Look around 15:23 for the deletion and re-entry of the new sensor manually. A AAPS restart didn't change the result on the UI.

[image]

MilosKozak, 2024-01-12:
this is question on xdrip how it does send the record

robertrub, 2024-01-12:
Yes, I wrote to them too and they already have a PR to solve the problem on their side.

What seems strange is that AAPS takes into account the xDrip data and not the one entered in xDrip. But, this may be explained by the date sent by xDrip being "newer"...

Keeping this one open till the xDrip PR gets merged and see what it does with AAPS. The PR should fix a similar problem with G7 too.

closed 2024-01-17 by MilosKozak