nightscout/AndroidAPS #3168: DynamicISF Errors when selected

ShusterRooster, 2024-01-10:
Hi AAPS Devs,

I haven't been able to use DynamicISF for many dev versions at this point, vanilla or modified. I can't pinpoint an exact version but the error seems to have started around November 2023. Whenever I try to launch DynamicISF, it simply errors and defaults to OpenAPS AMA. 

I have upgraded to the latest version to see if there has been any change but even on AAPS 3.30-dev the error persists.

https://github.com/nightscout/AndroidAPS/assets/66763392/94d12dde-c2fe-418e-953c-3adb956d0e26

Hopefully, the issue is portrayed well in the video. Please let me know if I need any other documentation. 

Here is the log file associated:
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13895203/AndroidAPS.log)

MilosKozak, 2024-01-12:
did you start Objective 11?

MilosKozak, 2024-01-22:
Closed for inactivity

closed 2024-01-22 by MilosKozak