nightscout/AndroidAPS #3172: Prevent translation for the abbreviation of hours and days

winni67, 2024-01-14:
In the master version of AAPS 3.2.0.3, there is a big mistake in the german translation. Someone changed the value
<string name="shorthour">h</string>
in the file
core/interfaces/src/main/res/values-de-rDE/strings.xml
to
<string name="shorthour">s</string>
But „s“ is the abbreviation for the physical unit „seconds“.
Now we see a lot of time information like
2d2s
instead of
2d2h
One hour (1h=3660s), thats a big difference and thats the mistake.
The problem for the german abbreviating/translation of day, hour, minute ans seconds are discussed here https://german.stackexchange.com/questions/14895/abbreviating-time

Longer abbreviations of days and hours in AAPS are not possible because of the problems on the overview page. There, the status line is interrupted and not shown completely, when the abbreviation of days and hours contain more than one character.

olorinmaia, 2024-01-14:
The best way to solve this is to join the translation team at crowdin and search for the string 'shorthour' and suggest the correct translation.

https://crowdin.com/project/androidaps/de

winni67, 2024-01-14:
@olorinmaia thats the way to correct that value. Thats already done. But we need a way to prevent this values (strings) from translating in the future.

olorinmaia, 2024-01-14:
For my part and as a translator of norwegian this is a string I want translated. I don't think a string can be made untranslatable for one language only. So with this example, and as with any other translatable strings, it's a matter of ensuring correct translation for given language imo :)

Maybe a discussion @ crowdin to make sure the right ppl approve translations?

MilosKozak, 2024-01-15:
there is a "short" in name. it means there is not much space for it

winni67, 2024-01-16:
The question is, if the error can be corrected in the master version. For me, I solved the problem by changing the code. I prefer day (d), hour (h), minute (min), second (s). Perhaps it could be possible to change the non standard units on the overview page (status line) by own local settings in the AAPS app. With big displays, people can use abbreviations with more than one character. On smaller displays, only one character for abbreviations is allowed. So, everyone can change it to the own preferred abbreviation.

MilosKozak, 2024-01-22:
all changes must go through dev

closed 2024-01-22 by MilosKozak