nightscout/AndroidAPS #3173: Error in dynamicISF algorithm

mordechi-david, 2024-01-16:
In version dev3.3.0 build on 14.01.24 the dynamicISF gives an abnormal result Attached is a picture I don't find the log something strange in this version I don't have recent logs
[image]

MilosKozak, 2024-01-16:
what is your dynisf factor? what it one time?

parapenT1sta, 2024-01-16:
@mordechi-david About the logs: https://github.com/nightscout/AndroidAPS/issues/3151#issuecomment-1873488139

parapenT1sta, 2024-01-16:
Same issue?
 https://github.com/nightscout/AndroidAPS/issues/3153

MilosKozak, 2024-01-16:
you had to set it back to previous value manually?

mordechi-david, 2024-01-16:
Yes, I restored it manually and it worked

MilosKozak, 2024-01-17:
I checked the code. The code itself only reads the value. So, to modify you have to:
* either enter preferences and change it
* or import another settings
If this is not your case you phone must be malfunctioning and I'd reset it to factory setting or replace at all

olorinmaia, 2024-01-17:
@MilosKozak It happend on my test-rig now after updating to latest dev. I haven't touched the setting ofc. attaching screenshots and logs. Can you check if there is something in the logs that can be helpful if this can be improved?

Probably around 14:00 - 14:10

[AndroidAPS_LOG_1705497291262.log.zip](https://github.com/nightscout/AndroidAPS/files/13963697/AndroidAPS_LOG_1705497291262.log.zip)

[image]
[image]

mordechi-david, 2024-01-17:
@MilosKozak I did change the preferences and it works fine
The only question is how it changed to 0 without me touching it

MilosKozak, 2024-01-17:
app was started at 14:08 and then restarted 
first run was already wrong
last run during night was ok
somthing happened between

idalioreis, 2024-01-20:
i think the dynamicISF factor is by default 0%, so cause this error, if update to value that i define it works.

olorinmaia, 2024-01-20:
> i think the dynamicISF factor is by default 0%, so cause this error, if update to value that i define it works.

default value is 100

`ApsDynIsfAdjustmentFactor(R.string.key_dynamic_isf_adjustment_factor, 100, 1, 300),`

idalioreis, 2024-01-20:
> > i think the dynamicISF factor is by default 0%, so cause this error, if update to value that i define it works.
> 
> default value is 100
> 
> `ApsDynIsfAdjustmentFactor(R.string.key_dynamic_isf_adjustment_factor, 100, 1, 300),`

i understand, but i never change this value, only choose dynamicISF and was 0%.  

i put 100% after, and work fine.

szantos, 2024-01-22:
I've entered preferences to check the value (did not edit it), and it was zero. I've set it back to the normal value, but 1 loop run was affected, and I had a similar high ISF. Unable to provide logs.

MilosKozak, 2024-01-22:
what was the value before?

szantos, 2024-01-22:
80

MilosKozak, 2024-01-22:
https://github.com/nightscout/AndroidAPS/commit/bc0eac0eb4900c401753cd514f3d2ea683e6de69

closed 2024-01-22 by MilosKozak