nightscout/AndroidAPS #3186: Self-Designated Maximum SMB Amount

Svagtlys, 2024-01-23:
This is an enhancement/feature request regarding how SMBs are delivered. 

The current SMBs are limited to the smallest of (with a personal example in parentheses):
- Current basal rate for next x minutes (.7u for 30 minutes)
- Half the amount of the required basal (sometimes up to 5u)
- The remaining amount needed to reach maxIOB that is set (Ours is set to 10u)

So, the largest SMB is about .7u for us, however even this can be too much to receive in one go right after a site change, causing unnecessary pain. We would like to request a way to manually set a maximum SMB amount (specifically, set a lower limit). With this option, we could perhaps use it in conjunction with automations (not sure, we haven't unlocked those yet) to limit SMB insulin amount and increase the delivery rate right after a site change.

MilosKozak, 2024-01-24:
this doesn't make sense much. what do you do if you need to send a bolus?

Svagtlys, 2024-01-24:
You mean how do we bolus for meal/correction? When using the original PDM with the Dash, we did extended bolus. Now with AAPS, our options are reduced as extended boluses aren't an option in looping, so we have to bolus, stop it partway when it hurts, and try to remember to bolus again in a few minutes. Rinse and repeat until the needed insulin is delivered (can be up to ten iterations of this).

We thought we may be able to use SMBs for this purpose once they were unlocked, but the lack of ability to set a lower maximum insulin amount means we have to disable SMBs on site change, do our "bolus, stop, try to remember to repeat", and then also try to remember to re-enable SMBs.

MilosKozak, 2024-01-25:
This doesn't have a good solution on AAPS (or any automated system) side if a SMB, which is pretty small, can be too much .... and algorithm needs to deliver needed insuling as fast as possible to be effective
I see this possible options:
* try another pump/insulin/infusion set
* lower the limit (from default 30 min) for SMB for borrowing insulin from the future. This will decrease SMB size
* disable issuing SMB at all. Then AAPS will use TBRs only. You should probably increase limits to have it more effective

Svagtlys, 2024-01-25:
We understand it's not ideal to not be able to inject all insulin at once. We'll depend on the basal limitation then.

closed 2024-01-25 by Svagtlys