nightscout/AndroidAPS #3192: Dev 3.30 latest DynISF error with activation

lostboy86, 2024-01-26:
[AndroidAPS_LOG_1706264093742.log.zip](https://github.com/nightscout/AndroidAPS/files/14064334/AndroidAPS_LOG_1706264093742.log.zip)
[image]

MilosKozak, 2024-01-26:
give me screenshots of your profile's isf

lostboy86, 2024-01-26:
Attach here the screenshots

MilosKozak, 2024-01-26:
nothing is visible. add it on web

lostboy86, 2024-01-26:
[image]
[image]

MilosKozak, 2024-01-28:
at what action this message appeared? is it replicable?

lostboy86, 2024-01-28:
I'm only turn on the DynISF on latest dev.

The banner appears a few moments after, then repeat itself

MilosKozak, 2024-01-29:
hmmm no data in logs. set logging levels to default and post new logs with timestamp

lostboy86, 2024-01-29:
How i can set logging levels to default?

MilosKozak, 2024-01-29:
there is a button on page where you set it

lostboy86, 2024-01-29:
On settings log, i have all checked.

lostboy86, 2024-01-29:
[AndroidAPS_LOG_1706547426289.log.zip](https://github.com/nightscout/AndroidAPS/files/14088184/AndroidAPS_LOG_1706547426289.log.zip)

MilosKozak, 2024-02-02:
can be resolved by removing APS resuts

closed 2024-02-02 by MilosKozak

lostboy86, 2024-02-02:
Sorry, how i can resolve?
The issues it's in the graph scale of IC, ISF, Basal.
DIA scale graph it's resolved.