nightscout/AndroidAPS #3193: Error bolus Calculator on latest dev 3.30 take IC /ISF on inexistent profiles Active

lostboy86, 2024-01-26:
I have only one profile named "FIASP" on nightscout and aaps latest dev.
When i launch the Bolus Calc, if i not select manually FIASP, takes value from not existent profile named Active, with IC and ISF different from mine.

Pay attention, with this bug many people can rise BG or experiment dangerous low BG.

[image]
[image]
[image]

robertrub, 2024-01-27:
Strange. I don't have your problem. <Active> and my active profile give the same numbers. Using latest dev-A.

[image]

Can you stop all downloads from NS and reactivate your profile in APPS. Then check again if you have the same problem.

lostboy86, 2024-01-27:
> Strange. I don't have your problem. and my active profile give the same
> numbers. Using latest dev-A.
>
> Screenshot_20240126_104919_AAPS.jpg (view on web)
>
> Can you stop all downloads from NS and reactivate your profile in APPS.
> Then check again if you have the same problem.

Yes i try now, also NS not had previous the Active Profiles, i have only
FIASP on ns and aps.

How i can delete the Active Profiles wrong?

lostboy86, 2024-01-27:
Now another user report to the FB group AAPS USERS same situation : 

[https://www.facebook.com/photo?fbid=3647468725537482&set=gm.3706718732882850&idorvanity=1900195340201874](url)

robertrub, 2024-01-27:
There is no profile named _Active_ . I'm almost sur that the profile download from NS creates the bug.

In AAPS settings, stop all downloads **from** NS. Keep only the uploads **to** NS.

Create a new profile with a new name in AAPS. Manually, copy the settings from your _FIASP_ profile to the new one. Activate the new profile.

Delete the _FIASP_ profile.

Check if you stil have the bug (difference between _Actual_ and _New_ profile when you use the bolus calculator.

lostboy86, 2024-01-27:
I also try this option.
Same bug.

BUT i found a strange solution.
If i deactivated autotune circadian on-off, automation profile change
on-off, now Active show same Value on calc from Fiasp.

Pay attention : above, in the post of fb user on aaps users, she had same
issue but with the latest master.
The master don't have obv autotune inside the app, like the dev.

In the previous version,i have no difference from active to my profiles, in
bolus calc.

MilosKozak, 2024-02-11:
I see on the backgroud screen  yellow background on profile name => you are running % change on profile. thus different values

closed 2024-02-11 by MilosKozak

lostboy86, 2024-02-12:
Thank you very much for help question!