nightscout/AndroidAPS #3196: 3.3 Dev À Widget shows ISF ??

robertrub, 2024-01-27:
Last Dev. I have a "normal" AutoSens information on the AAPS UI but a strange information on the AAPS widget. It is as if it was showing an ISF change but I don't use _Variable sensitivity_ or DynISF.

[image]

[image]

[image]

vanelsberg, 2024-01-30:
Confirming!

MilosKozak, 2024-02-01:
https://github.com/nightscout/AndroidAPS/commit/aa13cff67ea9c4ece56f3a4051711ee561b03daf

closed 2024-02-01 by MilosKozak