nightscout/AndroidAPS #3197: Yellow warning triangle when change a G7 sensor

MarkusKramer75, 2024-01-27:
Hello everyone, I have been using the Dexcom G7 for good 1 month now. I have noticed that every time I change the sensor, the yellow warning triangle remains next to the glucose value for more than 35 hours. Now I'm wondering if this is really necessary. There are no values for 30 minutes and then, in my opinion, this is not absolutely necessary. Perhaps this could be reconsidered and the time shortened? Personally, I think that would be a good idea. Thank you in advance for thinking about it. Best regards Markus

olorinmaia, 2024-05-10:
We have recently started using G7 as well and since each G7 sensor has it's own transmitter every sensor change will trigger this warning. G7 also seem to send BG at irregular intervals if a few readings are missed. Short story, this warning is visible almost all the time for G7.

This is only a warning, and AAPS operates as normal. But this could be improved in two ways the way I see it.

1) Consider not raising yellow warning sign when CGM = G7 and data gets recalculated.
2) Improve error message in AAPS so that users don't think this is possibly something that require action? I have improved it in Norwegian translation to something like this: 
"Recalculated data is used because some BS data is missing or that BS data arrives at an unexpected time. Will reset after about 24 hours. No action required"

Thoughts @MilosKozak ?

TWilhoWI, 2024-05-10:
Also got the issue!

Navid200, 2024-05-12:
@olorinmaia 
Thanks for mentioning this in the xDrip discussions.

I just did a test and reported the results here: https://navid200.github.io/xDrip/docs/Dexcom/G7OnceAMinuteScan.html

Hopefully, this change in behavior in G7 with respect to G6 will not require too much change in xDrip or AAPS.

Navid200, 2024-05-22:
After three consecutive missed handshakes (15 minutes), G7 moves into a once a minute transmission to help with the reconnect.
This change of behavior with respect to G6 confused xDrip.  
It was xDrip that screwed up the timestamps I am sorry about that.  But, this is fixed now.

If you are using xDrip to collect from G7, please use the May 22, 2024 release or any after that.  xDrip does not change the timestap any more under those conditions.

Every G7 (or G6) has a local free running oscillator.  There is no timing relationship, neither phase nor frequency, between the local oscillators of  different G7 devices.
Every time you start using a new G7, xDrip will adapt to its local oscillator.  Therefore, there will always be a shift in the tine stamp when starting a new G7.
There is nothing we can do to xDrip to change this behavior.

If you think there is anything else we can do, please let me know.

TWilhoWI, 2024-05-23:
[image]
Hi Navid,
Don't know if it is related, but got a couple of connection issues after upgrade. This is also a new sensor.

TWilhoWI, 2024-05-23:
Ow so the unknown packet is to be neglected?

TWilhoWI, 2024-05-24:
[image]
Still frequent connection drops. Do I need to provide more logs?

TWilhoWI, 2024-05-24:
Extra info.
[image]

TWilhoWI, 2024-05-24:
[image]

Navid200, 2024-05-24:
@TWilhoWI None of what you are reporting is related to AAPS.
This is the AAPS repository.

TWilhoWI, 2024-05-24:
Dammit, will repost it in G7 forum. Sorry

CAPTCG, 2025-01-13:
I am using an Eversense CGM. The BG readings come from the  Esel app to AAPS by MM640G CGM as selected in the configuration and are as good or better than those of a G6 or G7. So, why the yellow triangle "Recaluclated BG" with Eversense? This is not at all necessary. Can the yellow triangle resteiction please be removed from 3.3.1.0?

MilosKozak, 2026-10-11:
In the new user interface (current dev version), AAPS no longer shows a warning triangle when it uses recalculated BG data, for example after a G7 sensor change. A warning now appears only for doubled or flat BG data.

closed 2026-10-11 by MilosKozak