nightscout/AndroidAPS #3199: [Naming suggestion] Rename AAPS client app to AAPS Follower

yossibivas11, 2024-01-27:
I think AAPS client isn't a discriptive name and I have gotten feedback from several people that this name isn't underatood.
I suggest AAPS Follower.

olorinmaia, 2024-01-30:
Or even shorter name? AAPSFollow

It's amazing how much you can do in AAPSClient, it's simply much more than an app "following" AAPS as you can set TT, do profile adjustments, profile switches, register carbs, add manual insulin not delivered by pump, delete treatments, register notes and more. It's so powerful that you basically can run administer insulin directly via TT/profile switches, if SMB is setup correctly

AAPSFollow or AAPSFollower is though a more fitting name than AAPSClient. I struggle with coming up with any other alternatives :)

yossibivas11, 2024-02-07:
> > AAPSFollow or AAPSFollower is though a more fitting name than AAPSClient. I struggle with coming up with any other alternatives :)
> 
> AAPSControl ? Although that gives the illusion that you can also give bolus ?

I don't like it, it resembles pumpcontrol too much. AAPS Client doesn't control much anyway, most of its use is as a follower

gempickfordwaugh, 2024-05-28:
AAPS Supporter instead?

PieterDeBruijn, 2024-07-23:
Dexcom uses **Dexcom Follow**, so please use the same naming schedule. CamAPS has **CamAPS Companion**, that's also a good one!
"AAPS client" is not a good name in any case.