nightscout/AndroidAPS #3200: No profile set

xLimzy, 2024-01-28:
Hello! Just built the new version of Andriod Aps and imported my settings- but nothing is coming through- please help I've been without the closed loop for hours

lostboy86, 2024-01-28:
After import prefs and settings, have you stopped and reopen the app?

xLimzy, 2024-01-28:
Yes, I actually just noticed that on nightscout I am getting  "status: not loaded". No data there...?

Sincerely,

Lisa Coladonato

xLimzy, 2024-01-28:
Sorry, my nightscout is back up and running but my settings still haven't come through. Does exports from a different version transfer to updated versions?

Sincerely,

Lisa Coladonato

xLimzy, 2024-01-28:
Yes, nothing is coming through. Can exports from previous versions import to updated versions?

MilosKozak, 2024-01-29:
what do you see on nsclient tab?

lostboy86, 2024-01-29:
Nsclient From APS?

MilosKozak, 2024-01-31:
closed for inactivity

closed 2024-01-31 by MilosKozak