nightscout/AndroidAPS #3210: Easy to miss toggles on the calculator

Mendanbar, 2024-02-02:
[image]

Can we brainstorm some way to make the buttons a little more obvious in the hotspot shown?  Due to the proximity to the OK and Cancel buttons, the toggles are very easy to accidentally press, and when they are toggled only a dimming is applied on/off.  This makes it really hard for young or new users to the system to see, and it can often result in wildly incorrect bolus calculations.

Some ideas:
- Borders around the buttons (toggles and/or OK and Cancel)
- A more obvious on/off visualization
- Placement of the toggles moved to be harder to hit accidentally
- Confirmation dialog or some other mechanism to warn that the toggles have been pressed

Mendanbar, 2024-02-02:
An optional lock would be my preference I think. It could be something enabled in the settings and opt in, so that power users don't have new and cumbersome UI mechanics foisted upon them.

kenzo44, 2024-02-04:
1) This is one with more padding between the buttons and the cancel button
[image]

2) This is with padding and the icons are now placed inside buttons
[image]

Thoughts

Mendanbar, 2024-02-04:
@kenzo44 I really like the second one.  I wonder if it's possible to have the "toggled off" state appear without a button surround, and the "toggled on" appear as a raised button?  Visually that would make it very clear which are toggled on/off at a glance.

I think the extra padding will help to reduce accidental toggles.

I also still think that the safest approach would be to implement an optional lock to prevent any toggles without a pin entry or other "unlock" mechanism, but understand that it would represent quite a bit more work.

kenzo44, 2024-02-06:
I am on board with increasing the padding for now. 

We will need to discuss the button states - example, true state would mean pressed in, and false would be raised like a tape recorder.

I don't know about the optional lock mechanism you described. It might complicate the UX. At the moment, we have a 2-step confirmation before delivering insulin.

kenzo44, 2024-02-06:
I was thinking of the toggled states for the buttons and was not a fan of the raised look. I think this looks better and gets the point across.

[image]

I don't know about the optional lock mechanism you described. It might complicate the UX. At the moment, we have a 2-step confirmation before delivering insulin.

Mendanbar, 2024-02-06:
> I was thinking of the toggled states for the buttons and was not a fan of the raised look. I think this looks better and gets the point across.

I agree, this looks very clear and there's a nice contrast between the enabled and disabled state.

> I don't know about the optional lock mechanism you described. It might complicate the UX. At the moment, we have a 2-step confirmation before delivering insulin.

I think the core of the problem is that by the time the 2-step confirmation occurs, the status of these buttons may already have been missed.  It's more of a problem when passing on care to third parties (a school nurse, for example). 

One other potential solution might be to make the "buttons" in this collapsed (eye not checked) state only status lights.  In other words, what if you could only toggle the items if the eye was checked and the UI was in an expanded state?  That might require some more UI rework, as the expanded menu is not as easy to navigate or clear.  Just something to think about.

I appreciate your time, though, and I think just changing the button states to be more visually distinct is a huge improvement.

olorinmaia, 2026-09-23:
Closing, improved in AAPS 3.x and in dev AAPS v4

closed 2026-09-23 by olorinmaia