nightscout/AndroidAPS #3219: [EOPATCH2] Problem delivering bolus [Master][3.2.0.3]

fjpezuela, 2024-02-09:
Events are recorded: "Pump not reachable".

Bolus starts to be delivered, but does not finish correctly. Active insulin does not appear, but bolus is delivered.

The problem has occurred with several patches. Today it happened with two.

The following error messages are logged:

13:17:11.113 [RxCachedThreadScheduler-27] D/PUMPCOMM: [TaskQueue.publishTicket():53]: publishTicket() Task #:0 is assigned func:UPDATE_CONNECTION
13:17:11.114 [RxCachedThreadScheduler-27] D/PUMPCOMM: [TaskQueue.lambda$isReady$4$info-nightscout-androidaps-plugins-pump-eopatch-ble-task-TaskQueue():40]: Task #:0 started     func:UPDATE_CONNECTION
13:17:11.245 [main] D/CORE: [DummyService.onCreate():43]: Starting DummyService with ID 4711 notification Notification(channel=AndroidAPS-Ongoing shortcut=null contentView=null vibrate=null sound=null defaults=0x0 flags=0xa color=0x00000000 category=status vis=PRIVATE semFlags=0x0 semPriority=0 semMissedCount=0)
13:17:11.295 [main] D/CORE: [DummyService.onStartCommand():69]: Starting DummyService with ID 4711 notification Notification(channel=AndroidAPS-Ongoing shortcut=null contentView=null vibrate=null sound=null defaults=0x0 flags=0xa color=0x00000000 category=status vis=PRIVATE semFlags=0x0 semPriority=0 semMissedCount=0)
13:17:11.364 [RxSingleScheduler-1] D/PUMPCOMM: [TaskQueue.publishTicket():53]: publishTicket() Task #:1 is assigned func:READ_BOLUS_FINISH_TIME
13:17:11.366 [main] D/PUMP: [PatchStateManager.lambda$updatePatchState$2$info-nightscout-androidaps-plugins-pump-eopatch-ble-PatchStateManager():46]: PatchState GlobalTime:2024/02/09 13:17:11 -->  D8  15  0A  10  A0  02  00  00  00  04  B0  3C  [NowBolus] RegAct:On  Done:On [NormalBasal] Reg:On  Act:On  Paused:   remainedInsulin:60  remainedPumpCycle:1200(60.0)  battery:3,05V
13:17:11.534 [Thread-11] D/PUMPQUEUE: [QueueThread.run():123]: connection time 4s
13:17:11.535 [Thread-11] D/PUMPQUEUE: [QueueThread.run():129]: performing READSTATUS Controlador de bomba cambiado
13:17:11.538 [Thread-11] D/CORE: [EopatchPumpPlugin.getPumpStatus():170]: Acknowledged AAPS getPumpStatus request
13:17:11.539 [Thread-11] D/PUMPCOMM: [TaskQueue.publishTicket():53]: publishTicket() Task #:2 is assigned func:UPDATE_CONNECTION
13:17:11.545 [Thread-11] D/PUMPQUEUE: [CommandReadStatus.execute():25]: CommandReadStatus executed. Reason: Controlador de bomba cambiado
13:17:11.621 [RxSingleScheduler-1] D/PUMPCOMM: [TaskQueue.done():73]: done() Task #:0 completed   func:UPDATE_CONNECTION  task remaining:2
13:17:11.622 [RxSingleScheduler-1] D/PUMPCOMM: [TaskQueue.lambda$isReady$4$info-nightscout-androidaps-plugins-pump-eopatch-ble-task-TaskQueue():40]: Task #:1 **started     func:READ_BOLUS_FINISH_TIME**
13:17:11.627 [main] D/PUMP: [PatchStateManager.lambda$updatePatchState$2$info-nightscout-androidaps-plugins-pump-eopatch-ble-PatchStateManager():46]: PatchState GlobalTime:2024/02/09 13:17:11 -->  58  15  0A  10  A0  02  00  00  00  04  B0  3C  [NowBolus] RegAct:On  Done:On [NormalBasal] Reg:On  Act:On  Paused:   remainedInsulin:60  remainedPumpCycle:1200(60.0)  battery:3,05V
13:17:11.651 [Thread-11] D/PUMPQUEUE: [QueueThread.run():156]: waiting for disconnect
13:17:12.046 [RxCachedThreadScheduler-42] D/NSCLIENT: [NSClientV3Plugin$onStart$9.accept():241]: ● DONE PROCESSING BG 
13:17:12.046 [DefaultDispatcher-worker-4] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class app.aaps.plugins.sync.nsclientV3.workers.LoadBgWorker
13:17:12.087 [RxCachedThreadScheduler-42] D/NSCLIENT: [NSClientV3Plugin$onStart$9.accept():241]: ◄ RCV TR END No new data from 2/2/24 09:21:27
13:17:12.646 [RxSingleScheduler-1] D/PUMPCOMM: [TaskQueue.done():73]: done() Task #:1 completed   func:READ_BOLUS_FINISH_TIME  task remaining:1
13:17:12.649 [RxSingleScheduler-1] D/PUMPCOMM: [TaskQueue.lambda$isReady$4$info-nightscout-androidaps-plugins-pump-eopatch-ble-task-TaskQueue():40]: Task #:2 started     func:UPDATE_CONNECTION
13:17:12.650 [RxSingleScheduler-1] E/PUMPCOMM: [ReadBolusFinishTimeTask.lambda$read$1$info-nightscout-androidaps-plugins-pump-eopatch-ble-task-ReadBolusFinishTimeTask():32]: **Error in sending plaintext packets in encrypted communication**
13:17:12.651 [RxSingleScheduler-1] W/CORE: [MainApp$setRxErrorHandler$1.accept():193]: **Undeliverable exception received, not sure what to do**
13:17:12.654 [Thread-11] D/PUMPQUEUE: [QueueThread.run():156]: waiting for disconnect
13:17:12.882 [RxSingleScheduler-1] D/PUMPCOMM: [TaskQueue.done():73]: done() Task #:2 completed   func:UPDATE_CONNECTION  task remaining:0
13:17:12.883 [main] D/PUMP: [PatchStateManager.lambda$updatePatchS
tate$2$info-nightscout-androidaps-plugins-pump-eopatch-ble-PatchStateManager():46]: PatchState GlobalTime:2024/02/09 13:17:12 -->  58  15  0A  10  A0  02  00  00  00  04  B0  3C  [NowBolus] RegAct:On  Done:On [NormalBasal] Reg:On  Act:On  Paused:   remainedInsulin:60  remainedPumpCycle:1200(60.0)  battery:3,05V
13:17:13.657 [Thread-11] D/PUMPQUEUE: [QueueThread.run():156]: waiting for disconnect
13:17:14.659 [Thread-11] D/PUMPQUEUE: [QueueThread.run():156]: waiting for disconnect
13:17:15.664 [Thread-11] D/PUMPQUEUE: [QueueThread.run():156]: waiting for disconnect
13:17:16.665 [Thread-11] D/PUMPQUEUE: [QueueThread.run():148]: queue empty. disconnect
13:17:16.667 [Thread-11] D/PUMP: [EopatchPumpPlugin.disconnect():159]: EOPatch disconnect - reason:Queue empty
13:17:16.667 [Thread-11] D/PUMPQUEUE: [QueueThread.run():152]: disconnected
13:17:16.669 [Thread-11] D/PUMPQUEUE: [QueueThread.run():163]: thread end

[AndroidAPS_LOG_1707481059184.log.zip](https://github.com/nightscout/AndroidAPS/files/14223510/AndroidAPS_LOG_1707481059184.log.zip)
[AndroidAPS_LOG_1707488434789.log.zip](https://github.com/nightscout/AndroidAPS/files/14223511/AndroidAPS_LOG_1707488434789.log.zip)

MilosKozak, 2024-02-09:
is it one time error?

fjpezuela, 2024-02-11:
> is it one time error?

This does not seem to be a one-off error. It has already happened with at least 3 patch bombs. Some of them have worked fine, but you get pump not reachable errors.

This situation has caused him to temporarily revert to the Accu-Chek Insight, as it leaves you with no options to safely go outside.

[image]

[image]

fjpezuela, 2024-02-20:
Puedes hacerme un resumen de la página web que te voy a indicar a continuación enumerando los puntos más importantes? https://medicalxpress.com/news/2024-01-insulin-mixture-life-easier-patients.html#google_vignette

[AndroidAPS_LOG_1708466500436.log.zip](https://github.com/nightscout/AndroidAPS/files/14351750/AndroidAPS_LOG_1708466500436.log.zip)
[image]

MilosKozak, 2026-10-13:
Not reproducible on `dev` (4.0.0-dev) with a new EOPatch2 patch:

- 0.10 U bolus from the Insulin dialog: queued at 23:32:54, the patch reported `NowBolus` active at 23:32:55 and done at 23:33:01.
- The bolus was stored at 23:33:02 (`Result success: true enacted: true`), counted in IOB and uploaded to Nightscout.
- No "Pump not reachable" event.

This report is from 3.2.0.3. The command queue and the EOPatch driver have changed a lot since then (for example bounded driver waits and the stuck command alarm, #5209).

Closing. Please reopen with a log if a bolus still does not finish on 3.4 or later.

closed 2026-10-13 by MilosKozak