nightscout/AndroidAPS #3223: invalid pump history data detected. Open new issue and email log

alnassermahrouqi, 2024-02-16:
[image]

Had this error yesterday. I'm not sure what it means. My logs will be sent by email

MilosKozak, 2024-02-16:
you sent the logs too late (it contains only vere recent data).
Are you able to locate logs from mentioned time?

alnassermahrouqi, 2024-02-16:
I don't know if I can. I'll be on the look out if it happens again and submit a ticket and logs immediately. 

Al Nasser Al Mahrouqi

alnassermahrouqi, 2024-02-18:
I have resent the logs as quickly as I could, ill update the email with the request number

MilosKozak, 2024-02-20:
I didn't found the message in any of logs you sent

alnassermahrouqi, 2024-02-20:
Even in today's log? I mistakenly mentioned in my last log submission it happened at 3pm but it was 1pm

Al Nasser Al Mahrouqi

alnassermahrouqi, 2024-02-20:
1 31pm specifically. 

Al Nasser Al Mahrouqi

closed 2024-11-30 by MilosKozak