nightscout/AndroidAPS #3224: Duplicate IOB, COB and in treatments

gh4tf, 2024-02-16:
At 07:28 this morning, I gave 1u and announced 10g using the calculator. It definitely only gave 1u, but the treatments show 1u and 10g have been given twice and 2u and 20g were added to the IOB and COB respectively, although only 1u and 10g are showing on the graph.

Using v 3.2.0.2.

It is the first time we've noticed it.  It appears to have been fine since.  We're just concerned because we use SMS commands for our daughter when she's at school and if this happens again, AAPS may think she has double COB and give lots of SMBs, but also think she has more IOB than she does and hold off giving insulin.

I've located the log files under Maintenance, but is it too late to send them now?  If not, how do I send them please? There's a button to send them by email, but I didn't know if it sends them automatically when hitting that button, or if I need an email address.

Thanks in advance!
[image]
[image]
[image]

MilosKozak, 2024-02-19:
Send email works only when you use just after the issue appeared
Use this https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html to locate proper log from time of issue and add it here

gh4tf, 2024-02-19:
Thanks for replying Milos.

The duplicates occurred at 07:28.

The link you posted says to upload the file dated before the error occurred.  There is one zip file modified at 04:36 and the next one is at 7:38.  Should I unzip the one modified at 04:36 because this is before the error occurred, or the one from 07:38 please?

MilosKozak, 2024-02-20:
You can drop the zip here directly. Just verify wanted data is there
I need logs from some time before and after

gh4tf, 2024-02-20:
Thanks Milos.

Unfortunately, the earliest date showing in the logs now is the 17th February and it happened on the 16th :-(

MilosKozak, 2024-02-25:
create new issue with logs if you notice it again

closed 2024-02-25 by MilosKozak