nightscout/AndroidAPS #3226: [DEV] Error "Profile sensitivity value" is out of hard limits when changing APS

olorinmaia, 2024-02-18:
AAPS 3.3.0-dev-a
Build: 9c327b4e4c-2024.02.17
Logs: [AndroidAPS_LOG_1708209853078.log.zip](https://github.com/nightscout/AndroidAPS/files/14323055/AndroidAPS_LOG_1708209853078.log.zip)

Step-by-step to reproduce:
1. Start with "Dynamic sensitivity" disabled in SMB plugin. Run a loop or two.
2. Enable "Dynamic sensitivity". Execute loop, either on next loop run or manually via Loop plugin. Error should pop up.

Can't see any issues with profile values.

Screenshots:
[image]
[image]
[image]
[image]

Heiner1, 2024-02-18:
Same to me. Did you also switch from another AAPS-development-idea to the original dev-version?

Heiner1, 2024-02-18:
I'll try to reproduce it the same way.

Heiner1, 2024-02-18:
You're right, error message is reproducable, exactly the same behaviour. For me the difference is now: DynISF Loop is running again. Bevor I got more error messages and no falling back to normal SMB mode - loop had been deactivated. So I think it has something to do with database? 
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/14323536/AndroidAPS.log)

ga-zelle, 2024-02-18:
I came across that message a lot in the recent days. I was focused on some other topic and did not drill down to base reason. However, my impression was that the message text is very misleading because it seems related to variable_sens as used in DynamicISF and has nothing to do with the pump profile sensitivity. If not yet calculated then variabale_sens may get assigned 0.0 or 1.0 which triggers the message. If assigning 100.0 when not used or not yet calculated then there was no message.

vanelsberg, 2024-02-20:
> Step-by-step to reproduce:

Confirming,
Seeing the same on current dev#9c327b4e

lostboy86, 2024-02-20:
it's the same issue of mine : 

https://github.com/nightscout/AndroidAPS/issues/3192

robinsondan87, 2024-06-17:
Is there a way to work around this or stop the error from happening?

vanelsberg, 2024-06-18:
> Step-by-step to reproduce

Confirming. Have also been seeing this some time ago. Looks like your repro...

MilosKozak, 2024-11-13:
still there?

olorinmaia, 2024-11-13:
Yes. Do you need fresh logs?

olorinmaia, 2024-11-16:
Sharing logs and a screenshot. Same steps to reproduce as in original description of issue.

[AndroidAPS_LOG_1731769511492.log.zip](https://github.com/user-attachments/files/17787544/AndroidAPS_LOG_1731769511492.log.zip)
[image]

MilosKozak, 2024-11-27:
fixed in https://github.com/nightscout/AndroidAPS/commit/80fd9e4d8be5f72858954b491675bc92f14a07e7

closed 2024-11-27 by MilosKozak