nightscout/AndroidAPS #3235: AutoSens tuned ON even with settings bring OFF

robertrub, 2024-02-23:
I turned OFF AutoSens in the settings but it keeps going above 100%. Is there something I did wrong or is there a bug ?

[image]

[image]

[image]

olorinmaia, 2024-02-23:
There was a bug earlier, with Autosens not showing in options. If you update to latest dev you will be able to see both Autosens and DynISF-"Sens" (if Dynamic sensitivity is enabled).

MilosKozak, 2024-02-24:
check it on latest dev

closed 2024-02-24 by robertrub