nightscout/AndroidAPS #3236: Feature request: Add Units left as an Condition in automation

avalitan, 2024-02-23:
I want to have my phone notify me when I'm running low on insulin, multiple times.
(eg: 50U left, 25U left, 10U left, 5 U left & 0U left)
With automation's you would be able to add an arbitrary amount of notifications or alerts!

I'm an Omnipod Dash user and running:
AAPS Version 3.2.0.3
Build: bb1a276-2024.01.27
Flavor: fullrelease
Nightscout version: 15.0.2
Firmware Version: Firmware 4.10.0 / Bluetooth 1.4.0

jbr7rr, 2024-03-16:
Also mentioned in : https://github.com/nightscout/AndroidAPS/issues/2404

I'm working on this.

olorinmaia, 2024-03-18:
@MilosKozak you can close this issue due to https://github.com/nightscout/AndroidAPS/pull/3267

closed 2024-03-18 by MilosKozak