nightscout/AndroidAPS #3247: Xdrip sync plugin starts repeatly (every few hours) fullsync between pumpcontrol and xdrip

maskac, 2024-03-01:
Xdrip last nightbuild
virtualpump
BYODA G6 as source BG
Xdrip hw source: xdrip follower

Pumpcontrol 3.2.0.3
Build: bb1a276078-2023.12.17
Verze Nightscoutu: 15.0.2

[AndroidAPS_LOG_1709322758182.log.zip](https://github.com/nightscout/AndroidAPS/files/14466983/AndroidAPS_LOG_1709322758182.log.zip)
[AndroidAPS_LOG_1709323258734.log.zip](https://github.com/nightscout/AndroidAPS/files/14466985/AndroidAPS_LOG_1709323258734.log.zip)

MilosKozak, 2024-03-03:
It's probably db issue. I don't see any reason for such behavior and unable to replicate it. I'd recommend database reset

closed 2024-03-03 by MilosKozak

maskac, 2024-03-06:
Same issue few hours after DB reset :( 

[AndroidAPS_LOG_1709751815122.log.zip](https://github.com/nightscout/AndroidAPS/files/14514599/AndroidAPS_LOG_1709751815122.log.zip)