nightscout/AndroidAPS #3252: Latest Dev: Crash with Preference menu (Action Tab preference menu not disabled sometimes)

Philoul, 2024-03-02:
Action tab Plugin preference should normally be always disabled, but in some situation (another Tab on the right of Action tab with preference plugin enabled), we can have a refresh problem and plugin menu is enabled for Action tab, then if we select it AAPS Crash...

On screenshot below, 
To reproduce crash: Simple Mode disabled, I salected within OpenAPS SMB tab the 3 dots menu, then I switch back to Action Tab, and in 3 dots menu, Plugin preferences is enabled (a click on it Crash AAPS)
[image]

Philoul, 2024-03-02:
Note that if previous selected Tab was Overview, Action plugin Preference menu is disabled in 3 dots menu, but if previous selected tab is on the right of Action tab, plugin preference remains enabled within menu when I select Action tab...

MilosKozak, 2024-03-03:
fixed by https://github.com/nightscout/AndroidAPS/commit/c4cd90d69b53611407df2e118364623e9838e5d7