nightscout/AndroidAPS #3253: Add Automation Action for 'Wake and Tune Up' reilylink/orangelink/etc

dcacklam, 2024-03-02:
Using: Medtronic 722 pump & OrangeLink Pro

I often experience 'pump unreachable' errors wherein going to the 'Actions' tab and selecting 'Wake and Tune Up' fixes the issue.

The 'automations' tab offers an option to take action on 'pump unreachable', but does not allow 'Wake and Tune Up' as an action.

Adding this as an automation option would allow at least some pump-unreachable situations to self correct.

AngelAlex1975, 2024-05-06:
Using: Medtronic 722 pump & OrangeLink Pro

In my application ver AAPS 3.2.0.3 There is no trigger in AUTO  "pump unreachable" &  action type "Wake and Tune Up"
The ability to reawaken a connection when it is broken seems interesting to me.

MilosKozak, 2026-10-15:
Fixed in eecc385472 (dev). Instead of an automation action, the Medtronic driver now does this by itself.

**What was wrong:** when a tune-up failed (for example, the pump was out of range for a moment), the driver stayed in an error state. Every pump command waited 2 minutes and failed, and nothing tried to tune again. Only the "Wake and Tune Up" button, or a new connection to the RileyLink, fixed it. That is why the button helped.

**Now:** after a failed tune-up, the next pump command starts "Wake and Tune Up" by itself, at least 10 minutes after the failure. The loop sends a command every 5 minutes, so this happens without the user. While the pump stays away, it tries again every 10 minutes, not all the time, so the RileyLink battery is not used up by scanning.

**Tested on a real pump** (Medtronic 522 + RileyLink):
1. The pump was put in a metal box, and "Wake and Tune Up" failed.
2. The pump was taken out again.
3. Exactly 10 minutes after the failure, the driver tuned again and found the pump.
4. The loop set its temp basal.

Nobody pressed a button.

It will be in the next dev build.

closed 2026-10-15 by MilosKozak