nightscout/AndroidAPS #3256: Feature Request: time dependency for MAX BOLUS

ChristianeVie, 2024-03-03:
Dear team, 
as a parent of a DT1 kid, I think it makes sense to add a time dependency for max bolus.

The IC varies quite dramatically over 24h. I suggest to add a feature to have a max-bolus in relation to every entry in the IC table too. So every entry in the table should have a corresponding max bolus. That would lower the risk of my kid getting too much insulin in case I have only one max bolus per day available which corresponds to the insulin need for breakfast (the highest insulin need of the day) .

Thanks for thinking about it!
 I think this will help the parents which need to look always at nightscout that the kid did not bolus too much.

Regards, Christiane

AAPS 3.2.0.2
fullrelease

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

ChristianeVie, 2024-03-04:
Hi, thanks for the feedback. 
I have realized that max bolus = 10 in the morning makes troubles because it is way too much for the rest of the day. If I allow only 6 to  not have trouble during the day, then I have always to remind my kid to look a second time into the calculator in the morning for the remaining bolus. 
I agree, that this finding is based on the assumption, that the body can only handle a defined bolus amount per time of the day (morning more insulin, lunch time less insulin and evening in the middle =circadian) and that not everybody sticks to this.
But if people do not agree, than they can fill always the same max bolus :-) or we make it as not mandatory only as optional.
For me, it would be a safety increase and making kids more independent from their parents.

szantos, 2024-03-07:
The possibility to change preferences based on automations would solve this specific (and many other) need without complicating AAPS for all. The advanced feature of automations is only available after objective 10 - probably late enough to avoid novices messing it up.