nightscout/AndroidAPS #3259: Insight Bluetooth Pairing "Connection failed"

alfureu, 2024-03-09:
Hi all, not sure whether anyone is experiencing this problem. This morning I received a warning that my hone cannot reach my pump, tried repairing as I usually did before, but always getting "Connection failed" when the phone lists the Insight pump's bluetooth. It was working before without any issues

Android: GrapheneOS latest (build no 2024030800) with Android 14 on Pixel 7
AAPS: 3.2.0.4

alfureu, 2024-03-09:
Further details with logs: https://discuss.grapheneos.org/d/11383-request-for-testing-and-feedback-with-bluetooth-on-android-14-qpr2-grapheneos

Philoul, 2024-03-09:
Correct me if I'm wrong, but I understood this issue started after an update of Android phone with latest version of Android 14 with BT increase of security...
@MilosKozak , I have no skills in BT connexion myself to fix or test it, and I don't know if this kind of BT connexion issue could impact other pump drivers or only Insight driver...

alfureu, 2024-03-09:
> @alfureu Correct me if I'm wrong, but I understood this issue started after an update of Android phone with latest version of Android 14 with BT increase of security... 

Correct, but since this update is upstream to regular stock Android on Pixel 7 from Google so problems might be ahead for many users with this phone, I wonder whether anyone with a Pixel 7 can confirm such an issue with the QPR2 update

alfureu, 2024-03-11:
> I don't have Pixel to be able test it

It appears to be a bigger issue, that most likely affects not only the GrapheneOS ROM but stock Pixel devices post-QPR2 update. I suspect that this major security update that significantly affects bluetooth functions will be a problem on any other device with Android 14. More details about the discussions: https://discuss.grapheneos.org/d/11383-request-for-testing-and-feedback-with-bluetooth-on-android-14-qpr2-grapheneos

If I can provide any logs, please let me know, I am quite upset that this got broken, already thinking of buying another phone, which will be _very_ impractical

Jared-Is-Coding, 2024-03-13:
Just wanted to chime in with my experience here. Pixel 7a, Android 14: I've noticed that I get random bluetooth "hangs" where my phone will attempt to connect to the pump for an infinite amount of time until I cycle bluetooth off and on. Last occurrence was just last night for a 25 minute period.

Perhaps related, just wanted to keep the discussion going. Is there anything more specific I can do to help testing?

Thanks.

alfureu, 2024-03-13:
hm, most likely unrelated, had similar before my bluetooth connection worked, as I rotated sometimes during my sleep with the duvet, I might have weakened the BT signal to a degree that it disrupted the connection. 

The current issue is specifically related to a very recent Android QPR2 update which broke the BT connection between Pixel 7 and Insight pump. This security update comes to all the devices in the future...

Jared-Is-Coding, 2024-03-13:
> hm, most likely unrelated, had similar before my bluetooth connection worked, as I rotated sometimes during my sleep with the duvet, I might have weakened the BT signal to a degree that it disrupted [...]

I wonder if this would only cause issues with a _running_ session, but not affect future sessions started after the update?

Philoul, 2024-04-06:
@alfureu Do you know if there is any updates / fixes on android side on this topic?

alfureu, 2024-04-06:
> @alfureu Do you know if there is any updates / fixes on android side on this topic?

Oh yeah, as far as I understood the issue, apparently Google messed up with the bluetooth firmware/security patch tagging, the devs at GrapheneOS implemented it quickly from AOSP and it broke things. Since then they reverted the changes, so all good now. I am closing, and remain hopeful that this does not happen again. 

Thanks for the follow-up!

closed 2024-04-06 by alfureu