nightscout/AndroidAPS #3260: TDD not correct

bertroode, 2024-03-11:
Hi,
On march 9 I tried to invoke Dynamic ISF but then AAPS was switched back to OpenAPsAMA. To correct the problem I imported the reference settings to get back to the last known good. Subsequently I finished the objectives needed.
Today I noticed however that the TDD report is wrong sing march 9. See enclosed
[image]\

AAPS 3.2.0.3
build bb1a276078-2023.12.19

When you need any further information, please give me a message.

Best regards and thank for all the effort in impriving AAPS

Bert roode

MilosKozak, 2024-03-11:
This is not enough info to provide any opinion
BTW: you can reset precalculated TDD on Stats page

bertroode, 2024-03-12:
The information on the Stats page is the same as the data in my pump.

MilosKozak, 2024-03-13:
what pump is it?

MilosKozak, 2024-11-13:
closed for inactivity

closed 2024-11-13 by MilosKozak