nightscout/AndroidAPS #3263: AAPS 3.2.0.4 ComboV2 status tab unreadable in daytime display mode

NicoAAPS, 2024-03-12:
### AAPS version
3.2.0.4 
build  5d3e33ce8a
flavor fullrelease

### hardware and software
Samsung Galaxy A13 5G  - Android 14 - oneUI  6.0

### problem description

In daytime screen mode the status tab of the combo v2 has white text on white background, which makes it rather unreadable
Changing to night mode changes the screen into white text on black background which is quite readable.

### See attached screen shots
[image]
[image]

MilosKozak, 2024-03-12:
this is already fixed in dev
https://github.com/nightscout/AndroidAPS/pull/3258

MilosKozak, 2024-03-12:
just notice there will be only combo v2 driver in AAPS 3.3

closed 2024-03-12 by MilosKozak