nightscout/AndroidAPS #3264: History Browser date picker substracts one day (Android 14)

Jared-Is-Coding, 2024-03-13:
Hey everyone. Noticing that when browsing historical data using the History Browser from the preferences dropdown, the date selection menu will subtract one day from the highlighted day after pressing "OK".

Build: c31617fd2c-2023.12.15
Android Version 14 (UQ1A.240205.002, February 5th)

1. Open top-right dropdown
2. Select History browser
3. Select a date
4. Tap "OK"

What should happen: Date selected is presented
What actually happsn: One day prior to the selected date is presented

Let me know if further information is needed.

Example:
[image]
[image]

Jared-Is-Coding, 2024-05-13:
> Can't reproduce this problem on AAPS version 3.2.0.4 or current v3.3 dev builds. So, either the problem was fixed or what you are seeing is specific to your setup?
> 
> Suggest upgrading to AAPS release version 3.2.0.4 to see if this solves your problem.

Will attempt now.

vanelsberg, 2024-05-13:
> In addition to the initial bug report repro steps

Still no repro on my side (Samsung S22 Ultra, Android 14)
Did you check your phones date settings (region, Timezone, automatic date-time) anything unusual?

Jared-Is-Coding, 2024-05-13:
> > In addition to the initial bug report repro steps
> 
> Still no repro on my side (Samsung S22 Ultra, Android 14, TZ=CEST) Did you check your phones date/time settings (region, Timezone, automatic date-time), anything unusual?

Nothing out of place that I notice. The time of day has no effect on this selection bug, either, so I don't think it's a timezone thing.

happygirlnyc70, 2024-05-26:
I can confirm this behavior.on Pixel 4a 5G.  Also same thing happens with insert CGM see issue 3041. I'm currently on AAPS 3.2.0.4.

Knighth4wk, 2024-06-14:
I have this issue as well. It is happening everywhere the date can be picked. I have tested with date on history browser and careportal buttons on action tab.

Whaterver date I pick in date picker popup. After pressing OK, the date is changed to one day before.

Using Samaung S24 Ultra Android 14 with AAPS 3.2.0.4

MilosKozak, 2024-11-13:
What is your timezone/language setting? I cannot replicate the issue

Jared-Is-Coding, 2024-11-13:
> What is your timezone/language setting? I cannot replicate the issue

Timezone is EST/GMT-5. Language is en-US.

MilosKozak, 2024-11-13:
OK, I see the issue when I set GMT-5 on phone

Jared-Is-Coding, 2024-11-13:
> OK, I see the issue when I set GMT-5 on phone

Excellent. Any more information needed to reproduce, or do you have it happening consistently?

happygirlnyc70, 2024-11-13:
I am also GMT-5

MilosKozak, 2024-11-13:
it seems to be a bug in google's library. Reported https://github.com/material-components/material-components-android/issues/4373

MilosKozak, 2025-04-24:
workaround in https://github.com/nightscout/AndroidAPS/commit/50a3fe0f5ef367b1ad1c8cc700ab6d439623b217

closed 2025-04-24 by MilosKozak