nightscout/AndroidAPS #3265: Empty dialogue when logging CGM change via Action tab (minor cosmetics)

DM-LTR, 2024-03-14:
Issue: When user clicks the button for CGM Sensor insert, an empty dialogue appears and user has to select OK fot the second time.
I am really not sure if this is intentional (to somehow alert or delay the user and make him to click OK twice), but there was an extra confirmation message before (sth. like "Do you really want to record a sensor change"? OR maybe "This option will record the sensor change.")
--------------
Replication: Go to Actions tab, select "CGM Sensor insert" - 1st dialogue appears (Event date and time), OK, then **2nd one (empty)** appears, OK, done

https://github.com/nightscout/AndroidAPS/assets/43177472/dcba7d2e-9495-44e0-82de-4a1eb917fc32

jbr7rr, 2024-03-16:
Please include AAPS version in issue

Philoul, 2024-03-17:
On latest versions, it's not an empty dialog, you can tune date and hour of the event, and include a note (text information) (if Notes are enabled in Overview preferences)

DM-LTR, 2024-03-17:
> Please include AAPS version in issue

Sry. This issue is present for a long time. In this case master 3.2.0.3

MilosKozak, 2024-03-26:
https://github.com/nightscout/AndroidAPS/commit/cd5df709f6639ee7112ab62fda6e8778e0e71010

closed 2024-03-26 by MilosKozak