nightscout/AndroidAPS #3266: Enable SMB always while using xdrip in companion app mode and dexcom g7 original app

Pentagit, 2024-03-15:
When using xdrip (latest nightly) in companion app mode fetching bg values from dexcom g7 app, aaps still won't allow SMB to be enabled always. Please fix so that g7 users can use latest dexcom g7 app including clarity and still have hcl/fcl possibility in aaps.

jbr7rr, 2024-03-16:
Problem is that xDrip doesn't send which sensor the data comes from in companion app mode., so AAPS can't determine if it is data from G7 or e.g. a medtronic sensor
So we need this information from xDrip when we want to enable it for ONE/G7 companion mode, I suggest you open a discussion/issue over at xDrip as well.

Pentagit, 2024-03-19:
Thank you. I understand and will do that.
Though, why isn't AAPS enabling SMB always no matter which BG source xdrip is using?
AFAIK, xDrip companion mode is available for the following vendor apps:
G6 (master only)
One (only if the app shows BG notifications)
G7 (master only)
CamAPS
Guardian
MiniMed
Omnipod 5 app

Is any one of these apps a problem for always SMB in AAPS or why can't AAPS just enable it for the xdrip companion mode? But perhaps I'm misunderstanding something :)

jbr7rr, 2024-03-20:
> Though, why isn't AAPS enabling SMB always no matter which BG source xdrip is using? AFAIK, xDrip companion mode is available for the following vendor apps: G6 (master only) One (only if the app shows BG notifications) G7 (master only) CamAPS Guardian MiniMed Omnipod 5 app

For Dexcom we know that they have good filtering, meaning that when there are erroneous values the sensor/transmitter doesn't send them

Pentagit, 2024-03-24:
Why no mandatory smoothing (average/exponential) then when using xdrip companion data?

jbr7rr, 2024-03-25:
> Why no mandatory smoothing (average/exponential) then when using xdrip companion data?

Dexcom does error detection, meaning it will not send data when there are erroneous values., with smoothing only there will still be a datapoint on the erroneous value, which can lead to wrong dosing. But this is already being discussed in other issues as well:
https://github.com/nightscout/AndroidAPS/issues/3181
https://github.com/nightscout/AndroidAPS/issues/2121

MilosKozak, 2024-11-13:
Can anyone provide current state?

Pentagit, 2024-11-13:
Xdrip in companion mode still leads to

Pentagit, 2024-11-16:
I updated my AAPS app with androidstudio (fetch/pull/generate signed etc.) but somehow its the same build number just with a newer date:
Build: 3df83535f9-2024.11.16

Anyway it still does not work. 
Still: "Safety: SMB always and after carbs disabled because active BG source doesn't support advanced filtering." When chosing xdrip as bg source while xdrip getting values from dexcom app via companion mode .

closed 2024-11-27 by MilosKozak