nightscout/AndroidAPS #3268: [Medtrum Nano]: BT -> Connection timed out

dar-kod, 2024-03-17:
I started trying Medtrum Nano on 14 March at 18:52. I activated my first test patch and tried to give a bolus, but there was no connection for a long time. If I remember correctly, there was a "pump unreachable" notification. I restarted AAPS with Exit. That didn't help either. After about 10 minutes of waiting, a connection was made. But after about half an hour, the Medtrum took a few minutes to connect. I have used Dash before without any problems. I have a small problem with the Anubis transmitter, which often doesn't connect for 10-20 minutes, even when the phone is near me. It appears that disabling power optimization specifically for the Bluetooth system service on OnePlus devices is not directly possible through the standard user interface.

After that, the connection is usually established within 2 seconds. But sometimes, after receiving the BG reading, I see that it has been trying to connect for 120 seconds and still hasn't connected. With Dash, if I remember correctly, it always connected within 2 seconds or sometimes a little longer.

On 17 March at 1:56 and 2:05, I got "Connection timed out". I tried to give a small bolus 2 times with the message pump unreachable. It reconnected after I clicked on the three dots to exit. After a disconnection at 2:05, it didn't reconnect after the first exit but after the second. This is getting annoying.

AAPS version: AAPS 3.3.0-dev
Build: f2166ec366-2024.03.15
Phone: OnePlus 12
OS: OxygenOS 14.0
Bluetooth: 5.4, A2DP, LE, aptX HD, LHDC
Nano battery: 2.82 V

Whole log folder is zipped and attached.

[AAPS log files.zip](https://github.com/nightscout/AndroidAPS/files/14627504/AAPS.log.files.zip)

jbr7rr, 2024-03-17:
Thx for reporting, unfortunately the logs don't contain the time period of the connection issues, and no errors in the logs provided.

@MilosKozak Can we increase the log size? compared to master, dev logs are tiny

Can you setup the logs in the following manner, and upload logs when it happens again?
[image]

MilosKozak, 2024-03-18:
I didn't touch the size of logs

jbr7rr, 2024-03-18:
@dar-kod Can you test with https://github.com/nightscout/AndroidAPS/pull/3270 ?

Can you let it run for a couple of days and see if it fixes it? After that pls report back and send logs

dar-kod, 2024-03-19:
Thank you @jbr7rr. I started testing with #3270 at 23:47 the day before. No disconnects so far.

dar-kod, 2024-03-21:
@jbr7rr There have been no BT problems so far. I have yet to test if there will be when I apply a new patch. 

I have something strange and I am not sure if it is related to BT. When the bolus is visible in the progress bar, when I click stop it says "failed": "Bolus reported an error. Manually check real delivered amount". I have tried stopping 2 times today (15:42 and 17:35) and both times it was like this.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/14704626/AndroidAPS.log)

jbr7rr, 2024-03-22:
> @jbr7rr There have been no BT problems so far. I have yet to test if there will be when I apply a new patch.
>
> I have something strange and I am not sure if it is related to BT. When the bolus is [...]

This is expected behaviour when you cancel bolus

jbr7rr, 2024-03-27:
Merged in dev, can be closed imo. If you thing cancel bolus should not give a error please open new issue

closed 2024-03-27 by MilosKozak

fjpezuela, 2025-12-05:
@jbr7rr I'm adding new logs, from the first 30 minutes since the connection and the first unreachable pump error.

[AndroidAPS_LOG_1764935962652.log.zip](https://github.com/user-attachments/files/23960749/AndroidAPS_LOG_1764935962652.log.zip)

[image]
[image]