nightscout/AndroidAPS #3271: Omnipod Eros fails comm after 1d

frankaxela, 2024-03-18:
Since moved to 3.2.0.4 of AAPS, the pods are failing communication earlier than 3 days.
build: 5d3e33ce8a-28.02.2024
my setup is:
- Omnipod Eros
- Orange link pro with latest firmware
- xdrip+
- G6

[AndroidAPS_LOG_1710794405126.log.zip](https://github.com/nightscout/AndroidAPS/files/14642383/AndroidAPS_LOG_1710794405126.log.zip)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

vanelsberg, 2024-03-26:
Looks like comms problems (Phone-RL-device-Pod):
Next time this happens, try switching off and then on again your RL-device. If this does not solve restart the phone.
If the problem then still persists, try removing and re-registering your RL.

(Btw: always make sure you export AAPS settings for the currently active Pod so worse case you can reconnect to the active Pod when things fail and you need to reinstall)

vanelsberg, 2024-03-26:
If you are still using the original Rileylink device, consider switching to Orange link. It will give you much more stable connections and BT range...

frankaxela, 2024-04-01:
hi @vanelsberg thanks for the recommendations, I tried everything you mentioned for every pod that failed and nothing changed. Also, this issue didn't appear with the previous version of AAPS, 2.8.
These issues are happening with the latest.
I still think this is an issue

andyrozman, 2024-04-01:
If pods work in the beggining and they just run out of "juice" and are nor reachable anymore, then there could be 2 possible explanation:
1. We communicate to much which drains battery (which I don't think is the case)
2. Your pods battery has too low voltage for some reason (faulty pods or something)

If you compare code from 2.8 with 3.1, there were major architectural changes in whole application, so there could be million reasons... 

Now question is, do other Eros users encounter the same problems, if not, then this could be your local setup problem, although from what you have written, your setup seems to be standard one, that should work... Would it be possible that you have a series of faulty pods?

vanelsberg, 2024-04-01:
Yes. Agree.

> If pods work in the begining and they just run out of "juice" ...

ad2. Reason can be poor batteries are to old, which is the case when Pod's are beyond the expiration date.

frankaxela, 2024-04-03:
I thought that too, so I opened a different package and still had the same issue.

I will revert to the old algorithm as I changed to start using SMBs, maybe there's an increase in calls because of this too.

In the meantime, had another faulty pod, this time after 1d 20h. Which is early.
Tried the restart and disconnect the OL something is going on with the pods, will try to get some DASHs

[AndroidAPS_LOG_1712131187366.log.zip](https://github.com/nightscout/AndroidAPS/files/14851744/AndroidAPS_LOG_1712131187366.log.zip)