nightscout/AndroidAPS #3274: Feature Request: Insulin input should work even when no pod is active

andrejpodzimek, 2024-03-24:
With a closed loop setup (using Omnipod DASH and Dexcom G6), the UI has the following paradoxical limitation:

* **When a pod is active**, you can record “externally applied” insulin (e.g. a shot from a pen) whenever you want and the closed loop algorithm will take it into account.

  Even with an active pod, this is sometimes useful (albeit rarely), e.g. when you party so extremely hard that even 200 units for 3 days would not be enough. It’s of limited usefulness 99% of the time, but nice to have nonetheless.

* **When no pod is active**, you cannot record “external” insulin at all. The buttons that normally apply insulin are hidden.
  
  This is a severe limitation. It does happen (at least in my way of life) that a very long time (say, 2 hours) may elapse between pods. During that time, presumably, a pen comes in handy (to avoid a chat with death). Sadly there’s no obvious way (on the _overview_ screen) to record this “external” insulin and let the algorithm know about it.

olorinmaia, 2024-03-24:
I think this is coming for 3.3 master. Currently in dev. 
https://github.com/nightscout/AndroidAPS/pull/3254

andrejpodzimek, 2024-03-24:
Oh, I should have searched more thoroughly. This feature request is obsolete in that case. Thanks.

closed 2024-03-24 by andrejpodzimek