nightscout/AndroidAPS #3275: AAPS 3.3.0-dev-b: Administering carbs from watch tile are ignored by AAPS

vanelsberg, 2024-03-24:
**AndroidAPS 3.3.0-dev-b**
Build: 5b13a68d4d7076fdf695cd5115617f30ce982432
Watchface: same build, *AAPS Watchface: Custom Watchface*
Flavor: Fullrelease

**Problem description:**
Sinds latest dev build carbs entered from watch through the AAPS watchface tiles are only registered once after (re)starting AAPS. Any subsequent carbs are ignored by AAPS (not added to COB, not listed in treatments)

**Repro:**
See screenshots of AAPS treatments:
- AAPS restarted through ... menu:
At 22:05h carbs where administered from the tile for the amounts of 1,2 and 3 grams of carbs from the watch

- AAPS restarted through ... menu:
At 22:07h carbs where administered from the tile for the amounts of 11,12,13 and 14 grams of carbs from the watch

Note that only the entries for 1 and 11 grams (first actions after restarting AAPS) are registered. Anu subsequent carbs entered from the watch are ignored by AAPS. Standard confirmations on watch, no errors on watch er in AAPS

**Screenshots**
[image]

[image]

**logs & screenshots**
[image]

[AndroidAPS_LOG_1711314549045.log.zip](https://github.com/nightscout/AndroidAPS/files/14736884/AndroidAPS_LOG_1711314549045.log.zip)

vanelsberg, 2024-03-24:
The problem may have been introduced around commit 342141b3 at 18/03.
At least I noticed the problem for the first time around 19/03 after updating to latest build?

vanelsberg, 2024-03-24:
Remark:
Around the time administering the 1 grams of carbs through the watch I found following exception below.
*Note that the exception coincided with action administering 1 grams of carbs (which was succesfile!)(

**22:05:52.388** _ [RxCachedThreadScheduler-15] D/PUMPQUEUE: [CommandQueueImplementation.bolus_$lambda$10():273]: Going to store carbs_
_22:05:52.390 [RxCachedThreadScheduler-15] E/CORE: [FabricPrivacyImpl.logException():96]: Crashlytics log exception: 
java.lang.NullPointerException: null_
	at app.aaps.plugins.sync.wear.wearintegration.DataHandlerMobile.doBolus(DataHandlerMobile.kt:1286)
	at app.aaps.plugins.sync.wear.wearintegration.DataHandlerMobile.doECarbs(DataHandlerMobile.kt:1342)
	at app.aaps.plugins.sync.wear.wearintegration.DataHandlerMobile.doECarbs$default(DataHandlerMobile.kt:1334)
	at app.aaps.plugins.sync.wear.wearintegration.DataHandlerMobile$30.accept(DataHandlerMobile.kt:263)
	at app.aaps.plugins.sync.wear.wearintegration.DataHandlerMobile$30.accept(DataHandlerMobile.kt:261)
	at io.reactivex.rxjava3.internal.observers.LambdaObserver.onNext(LambdaObserver.java:63)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.drainNormal(ObservableObserveOn.java:202)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.run(ObservableObserveOn.java:256)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:307)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
	at java.lang.Thread.run(Thread.java:1012)_

olorinmaia, 2024-03-25:
Can confirm this issue myself in latest dev, I'm pretty sure this wasn't an issue when I tested this PR isolated directly from PR branch

vanelsberg, 2024-03-28:
Just tested:
commit#675fbb95d9f736bf5110e6b378e482c84ae2f827 (Merge PR:3254) was ok.
**Problem was introduced since** commit#f6ce817d423be7c3d56e8ab236b6f661e6b357a7 (fix merge errors for previous 2 commits on radicalb/dev_quick_wizard2)

@MilosKozak  Any ideas?

vanelsberg, 2024-03-28:
Btw: this morning same problem occurred with administering insulin from the tile....

Philoul, 2024-03-30:
@vanelsberg can you test my working branch shared on discord to check if issue is fixed or not?
- I fixed the exception you got in fabric privacy
- I made additional little update that could solve the problem

vanelsberg, 2024-03-30:
> @vanelsberg can you test my working branch shared on discord to check if issue is fixed or not?

Done.
Quick test shows this fixes the problem.
Thanks!

PR #3290 (2024-04-02): Fix Wear Tiles
Fix NPE
Reset lastBolusWizard and lastQuickWizardEntry on new preCheck (ECarbs, Bolus, ...)
Fix #3275
closed 2024-04-02 by MilosKozak