nightscout/AndroidAPS #3277: AAPS logfiles no longer available on users phone storage (e.g: /AAPS/logs)

vanelsberg, 2024-03-24:
**AndroidAPS 3.3.0-dev-b**
**Build**: 5b13a68
**Flavor**: Fullrelease

**Problem description:**
AAPS logfiles are no longer located at /AAPS/logs (nor on /android/infi.nighthscout...)
At least I could not find/access the logfiles on my phone from file explorer.

Note that AAPS does have logfiles:
When sending logfiles through e-mail (Manitenance menu) they are added to the e-mail.
logfiles .zip from e-mail is then available at a temporary **/AAPS/temp** location?

Philoul, 2024-03-25:
Log files has been moved to previous location (Android/data/...) but with Android 11, you can only access to log folder from computer (access to this folder is locked by Android).

vanelsberg, 2024-03-25:
> Log files has been moved to previous location (Android/data/...) but with Android 11, you can only access to log folder from computer (access to this folder is locked by Android).

I am running Android version 14, no logs for me at /Android/data:
- Files app: "_Due to Android restrictions, the content of this folder can only be shown on a computer_"
- File Manager+ app: **/Android/data/info.nighscout.androidaps i**s an empty folder
But you are right: when I connect the phone over USB I can see the logfiles at _\Android\data\info.nightscout.androidaps\files_

So to get to the log files I need to connect over USB or send e-mail. Imho not very convenient :-/
@MilosKozak For better user accessibility Suggesting to move log location back to the original location at /AAPS/logs?

vanelsberg, 2024-03-25:
> Log files has been moved to previous location (Android/data/...) but with Android 11, you can only access to log folder from computer (access to this folder is locked by Android).

I am running Android version 14, no logs for me at /Android/data:
- Files app: "_Due to Android restrictions, the content of this folder can only be shown on a computer_"
- File Manager+ app: **/Android/data/info.nighscout.androidaps i**s an empty folder
But you are right: when I connect the phone over USB I can see the logfiles at _\Android\data\info.nightscout.androidaps\files_

So to get to the log files I need to connect over USB or send e-mail. Not very convenient imho.
@MilosKozak For better accesebility Suggesting to move log location back to /AAPS/logs?

MilosKozak, 2024-03-26:
It's not possible because of newest Android limitations

closed 2024-03-26 by MilosKozak