nightscout/AndroidAPS #3278: AAPS 3.3.0-dev-b crashes when trying to use quick wizzard

cr582, 2024-03-25:
AAPS 3.3.0-dev-b crashes immediately as soon as you try to open the quick wizard.

cr582, 2024-03-25:
[AndroidAPS_LOG_1711366700050.log.zip](https://github.com/nightscout/AndroidAPS/files/14742840/AndroidAPS_LOG_1711366700050.log.zip)

Philoul, 2024-03-25:
I used quickwizard on my side to test it on latest dev without any crashes (within my dev configuration with android v11, Insight pump)

cr582, 2024-03-25:
I just double checked and deleted the old project, cloned it once again and made a clean build. AAPS immediately crashes as soon as I choose quick wizard. Samsung Galaxy S20 FE 5G Android 13

olorinmaia, 2024-03-25:
Can you try to go to settings > quick wizard > delete old QW ?

Add new QW and see if that helps? As a workaround.

cr582, 2024-03-25:
deleting the old QW entries and create new QW buttons did help indeed. thx

MilosKozak, 2024-03-26:
I don't see any crash around 12:32:30

MilosKozak, 2024-03-26:
I see crashes before but there is no stacktrace anyway.

closed 2024-03-26 by MilosKozak