nightscout/AndroidAPS #3279: 3.3 dev-b User entries duplicated for Carbs

Philoul, 2024-03-25:
As you can see in screenshot below, user entries are duplicated, first entry recorded within dialog (with correct icon) second entry recorded within PersistentLayerImpl (In think after analysis) with wrong icon as source:
Note that duplicated information is only for Carbs and is not for boluses...

[image]

I don't know the reason of this second entries within PersistentLayerImpl (but I think both could be removed in PersistentLayerImpl class because eather for a new entry or an update of an existing one, it should already be recorded with the correct icon where the action comes from (wear watch command, Carbs, Treatments, Calculator or QuickWizard dialogs, NS or SMS...)
[image]

MilosKozak, 2024-03-26:
I'd keep logging in PersistentLayerImpl only to make the logging independent to UI and other code

Philoul, 2024-03-26:
Ok, so I propose to use a dedicated icon / source for persistant layer (I suggest database icon), different than treatments icon used for Treatments Dialog in overview...

PR #3287 (2024-04-02): Create Database source for uel entry from CommandQueue
Fix #3279 to avoid misunderstanding with Treatments Dialog User entries

[image]
closed 2024-04-06 by Philoul