nightscout/AndroidAPS #3281: java.lang.NumberFormatException: empty String in Full Synchronization of Treatments

rvilchez1, 2024-03-25:
- Every time I perform a full synchorization I get an error when retrieving the Treatment data
- Build version: 3.3.0-dev-b

[AndroidAPS_LOG_1711403420056.log.zip](https://github.com/nightscout/AndroidAPS/files/14749996/AndroidAPS_LOG_1711403420056.log.zip)

Exception thrown:

22:49:19.357 [OkHttp https://ricky.ns.10be.de:15216/...] D/HTTP: [NSClientV3Plugin.setClient$lambda$6():344]: <-- END HTTP (249018-byte, 36028-gzipped-byte body)
22:49:19.370 [DefaultDispatcher-worker-1] E/CORE: [LoadTreatmentsWorker.doWorkAndLog():83]: Error: 
com.google.gson.JsonSyntaxException: java.lang.NumberFormatException: empty String
	at com.google.gson.internal.bind.TypeAdapters$7.read(TypeAdapters.java:251)
	at com.google.gson.internal.bind.TypeAdapters$7.read(TypeAdapters.java:241)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.readIntoField(ReflectiveTypeAdapterFactory.java:212)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$FieldReflectionAdapter.readField(ReflectiveTypeAdapterFactory.java:433)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:393)
	at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.read(TypeAdapterRuntimeTypeWrapper.java:40)
	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.read(CollectionTypeAdapterFactory.java:82)
	at com.google.gson.internal.bind.CollectionTypeAdapterFactory$Adapter.read(CollectionTypeAdapterFactory.java:61)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.readIntoField(ReflectiveTypeAdapterFactory.java:212)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$FieldReflectionAdapter.readField(ReflectiveTypeAdapterFactory.java:433)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:393)
	at retrofit2.converter.gson.GsonResponseBodyConverter.convert(GsonResponseBodyConverter.java:40)
	at retrofit2.converter.gson.GsonResponseBodyConverter.convert(GsonResponseBodyConverter.java:27)
	at retrofit2.OkHttpCall.parseResponse(OkHttpCall.java:243)
	at retrofit2.OkHttpCall$1.onResponse(OkHttpCall.java:153)
	at okhttp3.internal.connection.RealCall$AsyncCall.run(RealCall.kt:519)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
	at java.lang.Thread.run(Thread.java:1012)
Caused by: java.lang.NumberFormatException: empty String
	at jdk.internal.math.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:1842)
	at jdk.internal.math.FloatingDecimal.parseDouble(FloatingDecimal.java:110)
	at java.lang.Double.parseDouble(Double.java:660)
	at com.google.gson.stream.JsonReader.nextInt(JsonReader.java:1215)
	at com.google.gson.internal.bind.TypeAdapters$7.read(TypeAdapters.java:249)
	... 18 common frames omitted
22:49:19.371 [DefaultDispatcher-worker-1] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result FAILURE for class app.aaps.plugins.sync.nsclientV3.workers.LoadTreatmentsWorker Data {Error : java.lang.NumberFormatException: empty String, }
22:49:19.372 [RxCachedThreadScheduler-41] D/NSCLIENT: [NSClientV3Plugin$onStart$9.accept():249]: ◄ ERROR java.lang.NumberFormatException: empty String

MilosKozak, 2024-03-26:
It looks like misformatted json in NS. There is probably another app writing to NS

rvilchez1, 2024-03-26:
Only 2 instances of NSClient and one of AndroidAPS.