nightscout/AndroidAPS #3284: MEDTRUM Nano disconnecting

Korostensky, 2024-03-27:
After updating with https://github.com/nightscout/AndroidAPS/pull/3270, 
is the pump unreachable after an accidental disconnection (long phone distance) and no longer available.
--------------
Before the update, the phone (watch with full android) worked normally with the pump. 
I have been using it in this combination since 20.09.2023.
- 3.3.0-dev-b-app-full-release
- pump model - MD8201, FW 1.80.83
- phone model - KOSPET Optimus2, Android 10.7- non-standard, but fully functional until the last update.

To reconnect the pump, you must exit and restart the application. Then the pump will connect correctly. 

[image]
[Log Export (1).txt](https://github.com/nightscout/AndroidAPS/files/14771939/Log.Export.1.txt)

jbr7rr, 2024-03-27:
Interesting device..
How are other Bluetooth devices working? Other pumps, dash etc?

Before I make any modifications to the code, can you double check the following:
- Location permission set to "always allow"
- ANY powersaving measures for app, Bluetooth app. and general ?

Also can you send me logs which also contains successful connection ?

Korostensky, 2024-03-28:
I've been using different types of watches, customized with Nova Launcher, since 2020, the problem still remains battery life. So far I have only used it in combination with the DANA R and MEDTRUM Nano and the connection works quite correctly. Sometimes it happens that the signal from the Dexcom drops out, but it comes back on after a few minutes. The devices connected via Bluetooth are only the Dexcom G6 transmitter and pump, nothing else.  

All settings are correct, I have enabled constant location access, Bluetooth enabled in the app. I don't have the battery saver activated. Before the modification it worked correctly and when I install the older version, the pump disconnection does not happen.

Log attached. 
[Log Export-succesful.txt](https://github.com/nightscout/AndroidAPS/files/14785475/Log.Export-succesful.txt)

jbr7rr, 2024-04-20:
It looks your device needs to scan to properly wake up or reset the Bleutooth when a connection attempt fails.
However other devices don't handle scanning on every connection error too well. (see: https://github.com/nightscout/AndroidAPS/issues/3268)

I see few options

1. Add a retry mechanism and after 3 retries we start scanning anyway
2. Add a setting, for those phones who need it can enable scanning.
3. Add a check if android version is < 11 and do scanning for those devices. (This is a bit tricky as we only have 1 complaint so far with the new code)

@MilosKozak any thoughts ?

jbr7rr, 2024-05-10:
I found the probable cause for this issue, so no need for workarounds hopefully,

@Korostensky  Can you test with https://github.com/nightscout/AndroidAPS/pull/3331 ?

Korostensky, 2024-05-11:
@jbr7rr  Hi, unfortunately I sent the watch in for warranty, I'm currently using a Galaxy S23 phone. When it comes back I'll definitely try it out. Thank you very much.

Korostensky, 2024-05-27:
@jbr7rr  Hello, so unfortunately the pump does not connect after installing the current version and the problem persists. After accidental disconnection (long distance) the device in automatic mode no longer manages to connect to the pump.  I attach the current log.
[image]

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/15452014/AndroidAPS.log)

jbr7rr, 2024-05-27:
> @jbr7rr Hello, so unfortunately the pump does not connect after installing the current version and the problem persists. After accidental disconnection (long distance) the device in automatic mode no longer manages to connect to the pump. I attach the current log. ![20240527_085659](https://private-user-images.githubusercontent.com/54218454/333999171-47f2da8d-9916-45fe-b499-cf6878cadbb6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3OTQyNTYsIm5iZiI6MTcxNjc5Mzk1NiwicGF0aCI6Ii81NDIxODQ1NC8zMzM5OTkxNzEtNDdmMmRhOGQtOTkxNi00NWZlLWI0OTktY2Y2ODc4Y2FkYmI2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI3VDA3MTIzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM0MTMzZTFkZjYwYmNhMzIxMDMzZTRiOGEwMzBiZTA4NTk2NGE2M2NmMDliMWZhNDg1N2U5NjE1MzU1YjY3MzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.F2BZ6I7e37aWu_bd9ucLuBBiA6stjW9p_BhbYBOn9iE)
> 
> [AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/15452014/AndroidAPS.log)

Did you test with https://github.com/nightscout/AndroidAPS/pull/3331 ? This is not in dev yet, waiting for test results

jbr7rr, 2024-05-28:
> @jbr7rr thank you for the information. I didn't notice that 3331 is not yet part of the dev. I will wait.

Can you test the PR? I want to sure that it works, or that we need another solution, you are the only device that has this problem

Korostensky, 2024-05-28:
@jbr7rr I have a problem, the warranty repair failed and the device went to the dealer again for a warranty claim. It is a mechanical fault on the button. Perhaps not related to the quality of the product, but rather to the application of Murphy's Laws :).

When I receive the watch back, I will try to install the app with the code changes I see in the two files in 3331. 

I am not a programmer and do not know the mechanism of code changes in Android Studio, I can only change branches and merge changes across. But I simply change the lines of code as per the template in 3331 in my two files manually.

jbr7rr, 2024-05-30:
> @jbr7rr I have a problem, the warranty repair failed and the device went to the dealer again for a warranty claim. It is a mechanical fault on the button. Perhaps not related to the quality of the [...]

Hmm, I'll open the PR for merge, it has been tested enough so far on other devices

Worst case we might need a new setting after all for some specific cases, but we can see later.

Korostensky, 2024-07-04:
Hello, I got my watch back. I have installed the app built from the latest DEV and still getting the connection problem.
[image]

[AndroidAPS_LOG_1720095570457.log.zip](https://github.com/user-attachments/files/16098837/AndroidAPS_LOG_1720095570457.log.zip)

jbr7rr, 2024-07-13:
> Hello, I got my watch back. I have installed the app built from the latest DEV and still getting the connection problem. 

Seems we do need a setting to handle it properly for some devices.. bummer but ok :)

Can you test with:
https://github.com/nightscout/AndroidAPS/pull/3381
And enabling the new option that is present in the medtrum settings ?

Korostensky, 2024-07-15:
Thank you very much for this setup option. Unfortunately it's holiday time and it will take me a few weeks. I will give feedback when I can.

Korostensky, 2024-07-30:
Hi, I want to share my feedback. I have installed the application with your customized connection preset option.

[image]
[image]

I then deliberately lost the connection to the watch, and upon re-approaching, the Medtrum pump connected fine.

[image]

Thank you for this modification!

closed 2024-07-31 by MilosKozak