nightscout/AndroidAPS #3288: Use local http webserver to receive data

berek320, 2024-03-30:
Hi All, 
Is there any way in AAPS to receive data from the local Web server streaming carbs and insulin data?
This works perfect with xdrip just by listening on http://somekey@127.0.0.1:17580/api/v1/
But seems AAPS requires https to receive anything.
Any way to overcome this? Or patch an app any way to add this feature?
I don't need an online nightscout server, I just want to use this feature to not have to type in carbs&insulin twice in different apps.

Thanks

mick711, 2024-03-30:
I see this in pull request, but i think for security https is needed and now with v3 token. It was patched with this request but now don't work with new release.

MilosKozak, 2024-04-29:
this is very specific need. You must do it for yourself if you want it.

closed 2024-04-29 by MilosKozak