nightscout/AndroidAPS #3289: Feature request: Move automation user button next to treatments

jbr7rr, 2024-03-30:
I like to move the automation buttons next to the treatment buttons as this really cleans up the home screen, like this:
[image]

This works well with few buttons, however some users like to have a lot of automation user buttons and the result is hideous: 
[image]

I see 2 possible solutions for this:
1. A setting allowing to move the automation buttons next to the treatment buttons
2. When the automation buttons go over a certain threshold (e.g. 4 buttons) move it to separate row

Any ideas and/or comments are welcome

ga-zelle, 2024-03-30:
Just brainstorming:
- make the user buttons scrollable; may be easier to implement if they stay in their own space (c.f. scrollable list of actioned automations on the Automation tab page.
- drop the icon as it is the same for every button - or use first action icon as button icon

vanelsberg, 2024-03-31:
> I like to move the automation buttons next to the treatment buttons as this really cleans up the home screen

Well...
For me, this would only clutter the overview. Quite happy with the ways it is implemented right now.
Maybe also first ask users if they would like to change as is now?

What you are proposing would result in a lot of relatively small (hard to hit) buttons at the bottom of the screen.
Users won't be happy - or at least I would hate that ;-)

I could see one single "action" button that would open a modal/popup for me to select the desired action.

vanelsberg, 2024-03-31:
> Any ideas and/or comments are welcome

Question:
Why would I need buttons for manually operating "_automations_"?
Personally, I only see use in _automating_ based on specific events...

olorinmaia, 2024-04-08:
What about adding up to 4x manual triggered automations as a new tile in wear app? Ala how quick wizard tile work. That way it is optional also, and you can access them even easier if you use a watch.

koelewij, 2024-04-29:
I have conditional automation buttons that in some cases can show 6 buttons at the same time. That together with the 5 standard buttons on the bottom row do not really require this proposal. Only makes buttons too small. I can imagine some people would like to collapse all the buttons under one button, but personally I don’t need that, because now I have full overview. 
For the icons, I would suggest to remove them to reclaim space, or make them user defined. All the same icon/color is functionally useless.

olorinmaia, 2026-09-23:
@jbr7rr in dev (4.0.0) this has been changed so Automation / User actions appear next to Treatments button in the new Scenes button. Automations can also be added to QuickLaunch.

MilosKozak, 2026-10-11:
In 4.0.0 the automation user actions moved into the new Scenes/Automation button next to Treatments in the bottom bar. They can also be added to QuickLaunch. Please open a new issue if something is still missing.

closed 2026-10-11 by MilosKozak