nightscout/AndroidAPS #3295: Autosens value on AAPS Client never goes above 120% with 1.5 upper sensitivity ratio on AAPS Master

janvitos, 2024-04-05:
AAPSClient 3.2.0.4

When using an upper sensitivity ratio of 1.5 instead of the default 1.2 ratio with Autosens, the sensitivity shown on AAPS Client never goes above 120% even if it is 150% on AAPS Master (see photo below). It's as if AAPS Client is using the default upper sensitivity ratio of 1.2 instead of the custom chosen one in AAPS Master.

[image]

I wasn't able to test the custom lower sensitivity ratio, but the same issue might be present for that value as well. So if I set the lower sensitivity ratio to 0.5 instead of 0.7, then AAPS Client might never show anything below 70%.

MilosKozak, 2024-04-29:
It's can take up to 30 min to get min/max value sychronized to cleint

bjarnege, 2024-05-21:
Same problem here. Please let me know if there is anything else i need to check.

closed 2024-11-30 by MilosKozak