nightscout/AndroidAPS #3296: [Suggestion] export settings from maintenance go to dowloads folder or straight to google drive

Not-AriStienfeld, 2024-04-05:
Without a proper file sorting system, it can be annoying (and potentially confusing) to get the settings export onto the cloud. The option to send it straight to google drive would probably result in more people backing up more frequently, and would be more helpful for people who aren't as good at working with technology.

MilosKozak, 2024-04-29:
Storing config to some cloud is a way .... But changing diricetory will be impossible with upcomming Android versions anyway

vanelsberg, 2024-05-02:
Suggestion: I'm using FolderSsync to automagically sync AAPS settings the (cloud) storage of my choice.
(https://play.google.com/store/apps/details?id=dk.tacit.android.foldersync.lite&pcampaignid=web_share)

olorinmaia, 2024-05-11:
xDrip has automatic backup to google drive which is very useful

andyrozman, 2024-07-29:
@MilosKozak Could we add some sort of automatic backup like that? And we could see how Xdrip has this handled... My proposal would be to create export in such way that we just add date in readable format to the end (for example settings_20240702_1202.zip) and then copy it to specific google drive directory...
Is it even possible to create export now that we need password for creating export?

MilosKozak, 2024-07-31:
the problem is encryption of backup. AAPS doesn't know master password

KelvinKramp, 2024-08-01:
@MilosKozak  => maybe we can ask the user to input the password when it's time for a backup? 

I recently ran into the same issue. It has also been discussed in 2022 here: https://github.com/nightscout/AndroidAPS/issues/1208#issuecomment-1018023093

The options provided there were:
1. Automatically save settings on a regular basis (every day, keep last 5 days, last 4 nondays, last 12 months)
2. Automatically save settings after enabling a new omnipod (or other event requiring data from settings)
3. Provide an option to keep a copy of the settings in an alternate location
4. Provide an option to keep a copy of the settings in a predefined cloud location (note IMHO, since we are on android, google drive might be better suited than dropbox)

I would vote for a combination of 1 and 4 and for 4 (although more cumbersome) we could use mongodb via Nightscout for backupstorage
https://www.mongodb.com/developer/products/mongodb/storing-large-objects-and-files/

vanelsberg, 2024-08-01:
> the problem is encryption of backup. AAPS doesn't know master password

@MilosKozak As far as I can see the (sole?) purpose of encrypting the settings export file is to prevent sharing settings for the purpose of circumventing the need for acquiring the AAPS objectives. Right?

I will not share details, but most of you will know this can be overcome fairly easily without the need for any special tooling.
So I'd say the current solution of encrypting the settings on export (which required the users password) is not as sound as it suggests.

So to enable automatic backup of the settings using encryption the encryption key (e.g. password) need to be available to AAPS. Currently this is not the case and user is always asked to enter the password for exporting or importing settings.

*Proposal#1:*
_Securely store the users password in Android Keystore system._ This system allows you to store cryptographic keys in a secure container, making them difficult to extract from the device. Need to study details on implementing this but I think this is quite doable (Try Windows CoPilot for an example ;-)

*Proposal#2:*
_Generate an alternate encryption key based on proof of ownership._
One (sort of minimalistic) way would be to generate the key based on the Nightscout URL and API key configured for connecting the Nightscout site. Requirements:
1. Use the Nightscout connection specs as configured in AAPS
2. Require AAPS being successfully connected.

**Note**:  This solution is not fully secure. But would make it more complicated to share the encrypted settings file (at least for non-dev users)

**Proposal#1 would be my choice.**
Let me know your thoughts. I think I could do some draft on how to implement such solution?

vanelsberg, 2024-08-01:
@MilosKozak On the subject:
How about (optionally) storing the encrypted settings as a document in Nightscout. This would solve the users problem on making sure the settings are stored off-line (=not on the phone)?

andyrozman, 2024-08-01:
@vanelsberg Option 1 should be doable, there is simple Android way to Secure application settings and it should be pretty secure.

@KelvinKramp I looked at Google Drive API and is not very simple to use... You would expect that you could specify username and password (or even use settings directly from configured account on phone), but not such luck... 
I mean there must be a way to do this, I just haven't found it yet... 

Using Dropbox might be much easier, but I would like to avoid that way (person can have just 1 free Dropbox and if person is already is using it for some other stuff, using it for AAPS at same time would be a problem)... 

And using Nightscout ... I don't think we should use it... Lot of people stopped using it...

but maybe someone has some other idea... Someone mentioned that XDrip is using Cloud for backup, maybe we should look into that.

vanelsberg, 2024-08-01:
> @vanelsberg Option 1 should be doable...

Ok, I'll have a look at that then.

vanelsberg, 2024-09-25:
> @vanelsberg Option 1 should be doable...

Delay because of personal/family priorities. Just started some work on this. will try some draft/concept and when I like it, go for a draft/PR for dev's to review.

vanelsberg, 2024-09-29:
After looking at Android Keystore I must conclude that this does not solve. It is great for storing secret encryption keys but to decrypt an encrypted value it needs a .... "password" :-|

I think the AAPS master password is mostly needed for preventing users to export and share their AAPS settings which include objectives. Will not go into details, but most dev will know there are easy solution for doing this without the need for knowing the password ;-)

So I decide for a solution where the password entered by the user on settings export is stored on the phone's android datastore (which should be safe for this purpose). If the stored password is not used for export within N days (default=7) it is removed from the store.

Now have a working implementation for exporting AAPS settings through (scheduled) automation. Dev should be able to build on this f.i. for exporting to Cloud or triggering exports on pump-activation.

Planning to do a PR on dev for this week.
I'd prefer some experienced dev to do some review on the implementation. _Any volunteers?_

[image]

[image]

olorinmaia, 2024-10-01:
@vanelsberg Thanks for looking into this! :) Sounds like you are making great progress here on this very useful feature request. In dev there is an automation rule now allowing for certain events to happen if canula age is x hours etc. So, one could say if 0 hours run backup automatically. Since xDrip has a working automatic backup towards Google drive, maybe that's the best place to start?

I can surely help with reviewing/testing PR.

vanelsberg, 2024-10-02:
@olorinmaia Tnx!
Currently testing final (DRAFT) PR. Expect soon.

PR caches password on local phone. Stored pass word will expire when not used.
Still thinking about maybe a follow-up on the need to somehow encrypt the (without needing another password/key :see_no_evil: )

vanelsberg, 2024-10-02:
https://github.com/nightscout/AndroidAPS/pull/3451

vanelsberg, 2024-10-02:
Additional note:
This PR should enable exporting to alternate storage locations (like the Downloads folder or Cloud)
But this will be subject for another PR.

olorinmaia, 2026-09-23:
https://github.com/nightscout/AndroidAPS/pull/4481 & https://github.com/nightscout/AndroidAPS/pull/3487 is merged, closing.

closed 2026-09-23 by olorinmaia