nightscout/AndroidAPS #3297: AAPS turns off Bluetooth

Duester64, 2024-04-07:
Version: 3.3.0-dev-b
Phone: S21

Found this behavior during the last 6 weeks. At the beginning bluetooth was suddenly turned off a few times a day only, later this happend more often untill today were bluetooth can't be switched on anymore on the Handy untill AAPS  is deinstalled and started again. Now again it starts to switch off Bluetooth a few times a day. When it happens also the BYODA stops working.

BYODA is running flawless alone without any other BT connections, so it is possible that AAPS will cause this error. I'm wondering because the number of errors seems to accumulating day by day until BT is finally totally blocked. 

Something within AAPS seems frequently try to stop BT which will end in an command overflow which can't be cleared by the software. It can't be started then manually neither.

I deinstalled AAPS and BT working fine all day until AAPS is installed again. Dexcom soft is workiing fine alone also.

Hope this helps to solve this issue.

jbr7rr, 2024-04-07:
Turn of BT watchdog in AAPS and see if that fixes it and that your pump keeps.woeming reliable.

Also: add logs and specify pump type

MilosKozak, 2024-04-08:
If you turn off th option in AAPS, it's not commin from AAPS. Maybe xDrip?

vanelsberg, 2024-04-10:
> It worked flawless almost 2 yyears. A few weeks ago suddenly BT turned off without any reason from time to time.

Did you check your phones battery saving options?

Duester64, 2024-04-10:
I have absolutely no idea why this happened. I changed to a fresh phone now. Luckily my company send this offer. Now it's working fine on a S23. It seems that the old S21 phone had a problem.

closed 2024-04-29 by MilosKozak