nightscout/AndroidAPS #3298: Data does not upload to Nightscout after switch the Insulin type to Lyumjev

amitayg, 2024-04-09:
**Scenario:**
We use Omnipod Dash with Dexcom G6 
We moved to use Lyumjev Insulin 2 days ago 
In the configuration builder we configure the AAPS in the "Insulin Type section to use Lyumjev instead of "Rapid Insulin Oref.." 
Since we did that , the child's phone does not upload data to Nightscout and we also did not see his sugar data in the Dexcom Follow app either
Everything is fine with the child on the phone, he sees sugar data in Dexcom and in AAPS and also receives basal etc..
However , we could not see any data in NightScout site.
In addition we could not see any data in the Dexcom Follow app

**The Bug:**
It turns out that if you start using Lyumjev and change it in the AAPS configuration builder as well, it make problem in the pump and the loop and the AAPS does not upload data to the cloud and the Dexcom also stops showing data in the Follow app
Only After we switched back to "rapid insulin" everything worked out and we could see data in Nightscout site and in the Dexcom Follow app

MilosKozak, 2024-04-09:
There must be a different reason. Uploading to Dexcom has NO relation to AAPS

vanelsberg, 2024-04-10:
> There must be a different reason. Uploading to Dexcom has NO relation to AAPS

Confirming for dev-x. Have been switching insulin type several times in the past view weeks. No issue...

amitayg, 2024-04-10:
Note that this comes together with the clock change (move to summer
clock)in addition the child phone is with Google parental control
(Family link).
i think that the combination of the above cause the issue :

Scenario:
1.Change the insulin type
2.phone under parent supervision with Google Family link
3.Clock should be changed to summer clock (one hour forward)
4.before the child go to sleep i disabled the automatic time update in the
android configuration
5.the next day i move the clock manually 1 hour forward to summer clock

From this stage we have started to get many problems and error messages
that the pump is not synchronized and no basal will be received etc..
We played a lot with the phone time and update the pump accordingly
Eventually , no data uploaded to Nightscout unless we change the insulin
type to Rapid again and from this stage everything fixed

MilosKozak, 2024-04-29:
Manually setting time can cause lot of troubles in synchronization. You should avoid it

closed 2024-04-29 by MilosKozak